wpilibsuite/allwpilib #6312: Can't construct a Rotation2d from a `units::turn_t` in C++

auscompgeek, 2024-01-26:
**Describe the bug**

The C++ `Rotation2d` constructor doesn't accept a `units::turn_t`. The compiler complains that the overloaded call is ambiguous.

```
build/temp.linux-x86_64-cpython-310/gensrc/wpimath_geometry/Rotation2d.cpp: In lambda function:
build/temp.linux-x86_64-cpython-310/gensrc/wpimath_geometry/Rotation2d.cpp:321:28: error: call of overloaded 'Rotation2d(units::angle::turn_t&)' is ambiguous
  321 |     return Rotation2d(value);
      |                            ^
In file included from /home/davo/dev/frc/robotpy/mostrobotpy/subprojects/robotpy-wpimath/wpimath/_impl/include/frc/geometry/Rotation2d.h:181,
                 from build/temp.linux-x86_64-cpython-310/gensrc/wpimath_geometry/Rotation2d.cpp:8:
/home/davo/dev/frc/robotpy/mostrobotpy/subprojects/robotpy-wpimath/wpimath/_impl/include/frc/geometry/Rotation2d.inc:19:11: note: candidate: 'constexpr frc::Rotation2d::Rotation2d(units::angle::degree_t)'
   19 | constexpr Rotation2d::Rotation2d(units::degree_t value)
      |           ^~~~~~~~~~
/home/davo/dev/frc/robotpy/mostrobotpy/subprojects/robotpy-wpimath/wpimath/_impl/include/frc/geometry/Rotation2d.inc:14:11: note: candidate: 'constexpr frc::Rotation2d::Rotation2d(units::angle::radian_t)'
   14 | constexpr Rotation2d::Rotation2d(units::radian_t value)
      |           ^~~~~~~~~~
```

**To Reproduce**

```cpp
frc::Rotation2d FromRotations(units::turn_t value) {
  return Rotation2d(value);
}
```

**Expected behavior**

It should compile.

calcmogul, 2024-01-26:
Removing the units::degree_t overload may fix it, and shouldn't break anything.

auscompgeek, 2024-01-26:
It looks like the overload was added in #2564. I'm guessing there is now an implicit conversion between unit types of the same category (I suppose as evidenced by this issue), so it's no longer an issue?

auscompgeek, 2024-01-26:
Apparently that doesn't work - see build logs in #6314 for more.

clangd gives me this in [`TrajectoryGenerator.cpp` line 33](https://github.com/wpilibsuite/allwpilib/blob/68736d802d0764d8116dfc82fe42819b0cdbbbc3/wpimath/src/main/native/cpp/trajectory/TrajectoryGenerator.cpp#L33) when I delete that overload:

```
Transform2d.inc(15, 24): Candidate constructor not viable: no known conversion from 'angle::degree_t' (aka 'unit_t<unit<ratio<1, 180>, unit<ratio<1>, base_unit<ratio<0L, 1L>, ratio<0>, ratio<0>, ratio<1>>>, ratio<1>>>') to 'Rotation2d' for 2nd argument
```

PR #6316 (2024-01-26): [wpimath] Make Rotation2d implicitly convert from any angle unit
Fixes #6312.
Supersedes #6315.
closed 2024-01-26 by PeterJohnson