wpilibsuite/allwpilib #6330: Rumble In Simulation

levyishai, 2024-01-31:
**Is your feature request related to a problem? Please describe.**
While running a simulation through vs code and connecting to the Driver Station, rumble doesn't work for XboxController.

**Describe the solution you'd like**
Piping rumble through the driver station (or possibly making rumble work without even connecting to the driver station)

Starlight220, 2024-01-31:
Does this work in simulation with the DS plugin?

levyishai, 2024-01-31:
> Does this work in simulation with the DS plugin?

It doesn't

PeterJohnson, 2024-01-31:
Due to limitations in the underlying libraries we use, it will take a fair amount of work to implement this in the sim GUI DS.  It should work with the real DS though.  Does rumble work when connected to a robot?

levyishai, 2024-01-31:
I re-tested and rumbling works 
Not sure what I did wrong last time

closed 2024-01-31 by levyishai

fangeugene, 2025-03-27:
Sorry for resurrecting an old issue. @PeterJohnson should I expect the rumble to work in simulation if "Use Real Driverstation" is selected, or does your response above apply to this use case as well? Thanks!

ThadHouse, 2025-03-27:
The Real DS sim plugin should allow rumble to work. Although it hasn't been tested in a while.