wpilibsuite/allwpilib #6334: Method descriptions for all of the SmartDashboard.setDefault* methods are wrong

chauser, 2024-02-01:
The comments on, and therefore the class documentation descriptions, for all of the `SmartDashboard.setDefault*` (Java) say `Gets the current value in the table, setting it if it does not exist.`, but in fact these methods all return a boolean so they cannot be returning the value from the table. The documented behavior would be pretty useful, but I understand the need to have an indication if the default value could not be set because the table key exists with a different type. 

Documentation for the analogous methods in C++ has the same issue.

sciencewhiz, 2024-02-04:
Looks like the docs have been wrong ever since the functions were implemented 8 years ago. https://github.com/wpilibsuite/allwpilib/commit/64ebe7f5e50cf756d8621eec05cb6a30f8b362b7#diff-65701be4c8a216555abf8bf0942e1da98b20fef3a0ad44f3095f6d398bd296ebR294

rzblue, 2025-10-03:
Resolved in #6490

closed 2025-10-03 by rzblue