wpilibsuite/allwpilib #6336: Simulation consumes all CPU

truher, 2024-02-01:
When I run simulation of our Java robot on my Linux machine, it pegs all the cores.

VisualVM says the process isn't doing much except waiting on notifiers.  Maybe those are spinning?

It doesn't seem to hurt anything unless I want to do anything else, like run AdvantageScope.

My machine is Ubuntu 22.04, I'm using 2024.2.1.

PeterJohnson, 2024-02-01:
The Simulation GUI is a single thread but runs continuously—you can set a max frame rate in the GUI to try to help with that.  I’m not sure why you’re seeing all cores very active, as you’re right, things should be waiting on notifiers.

truher, 2024-02-01:
we have rev, phoenix, navx, and choreo, but none of that is used for simulation, we use simple fakes in java-land.

turning the frame rate down from 60fps to 30fps reduces the load from 4 cores to 3.

i guess looking more closely java isn't consuming *all* CPU, just most of it (there are 6 cores on this machine).

also i think my concern about advantagescope might be more about the 3d field implementation?  it goes at like 2 fps.