wpilibsuite/allwpilib #6349: [Feature] Accept PIDConstants when constructing a PIDController

GalexY727, 2024-02-06:
Don't you love having all those pesky PID Constants in one type? It's all good and great until you want to make a `PIDController` and realize that the `PIDConstants` you made aren't compatible with the controller. 

This feature could also possibly extend into the `HolonomicDriveController` which only accepts `PIDController`s in its constructor when it might be possible to make those `PIDControllers` inside of the constructor when given just `PIDConstants`.

spacey-sooty, 2024-02-06:
Does wpimath have a PIDConstants object or is this something we would need to add?

Starlight220, 2024-02-06:
It does not.

GalexY727, 2024-02-06:
It does 👀

spacey-sooty, 2024-02-06:
I think a PIDConstants class could be a nice addition but it'd be different to the pathplanner lib one. We'd probably want to move the IllegalArguments to there etc

TheTripleV, 2024-02-06:
I don't thing this needs to be a thing. Special casing arguments as a single type for constructors is unnecessary over abstraction. Imo this is along the lines of going too far a la pidcommand and pidsubsystem.

TheTripleV, 2024-02-17:
Going in this direction will make WPILib less maintainable and lead to more code breakages in the future as APIs evolve. I would recommend wrapping PIDController to add custom functionality for your team's specific use case or just continuing to pass in all of the arugments as is.

closed 2024-02-17 by TheTripleV