wpilibsuite/allwpilib #6371: 2024-crescendo.json error?

truher, 2024-02-15:
in the json file

https://github.com/wpilibsuite/allwpilib/blob/main/apriltag/src/main/native/resources/edu/wpi/first/apriltag/2024-crescendo.json

the speaker tags on the red end (e.g. 4) have x dimension 16.579.  the speaker tags on the blue end (e.g. 7) have x dimension -0.038, "behind the glass" i guess?

the json file specifies the field length as 16.451.
the game manual specifies the field length as 16.54.

using the game-manual field length yields consistent dimensions on red and blue sides -- in both cases the tag is about 0.038 behind the glass.

the json field-length doesn't yield consistent numbers, the red side is 12.8 cm behind the glass.

is this a simple transposition error?

calcmogul, 2024-02-15:
The JSON used a field length between the field sides of the alliance walls. The game manual may use something different. I had to ask Kevin to measure it for me from the field CAD.

Kevin-OConnor, 2024-02-15:
Yep, I must have transposed when typing it into the message. I see 16.541 in the CAD

PR #6373 (2024-02-15): [apriltag] Fix field length in 2024 JSON
Fixes #6371
commit d4d0545d (2024-02-15): [apriltag] Fix field length in 2024 JSON (#6373)

Fixes #6371
closed 2024-02-15 by PeterJohnson