wpilibsuite/allwpilib #6372: Sequential Command Fails Assertion when Executed

MrRedness, 2024-02-15:
**Describe the bug**
Whenever a sequential command is executed (only tested during simulation), the program fails the following assertion
`frcUserProgram: /work/wpiutil/src/main/native/thirdparty/llvm/include/wpi/SmallVector.h:296: T& wpi::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::operator[](wpi::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::size_type) [with T = std::unique_ptr<frc2::Command>; <template-parameter-1-2> = void; wpi::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::reference = std::unique_ptr<frc2::Command>&; wpi::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::size_type = long unsigned int]: Assertion idx < size() failed.`

**To Reproduce**
Steps to reproduce the behavior:
1. Create a sequential command group using provided factories (either frc2::cmd::Sequential or the .AndThen() decorator)
2. Try executing said command, either by scheduling it or with a simple .Unwrap()->Execute()
3. Exception is thrown

**Expected behavior**
No exception should be thrown, command should successfully execute

**Screenshots**
(This was put in RobotPeriodic())
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10, POPOS/Ubuntu 22.04 (same assertion failure on both)
 - Project Information: 
Project Version: 2024.2.1
VS Code Version: 1.83.1
WPILib Extension Version: 2024.1.1-beta-4
C++ Extension Version: 1.17.5
Java Extension Version: 1.23.0
Java Debug Extension Version: 0.52.0
Java Dependencies Extension Version 0.23.0
Java Version: 17
Java Location: /home/joseph/wpilib/2024/jdk
Vendor Libraries:
    PathplannerLib (2024.1.6)
    CTRE-Phoenix (v6) (24.1.0)
    WPILib-New-Commands (1.0.0)

**Additional context**
Parallel command works as indented

MrRedness, 2024-02-15:
I believed I localized the problem. The m_currentCommandIndex variable in SequentialCommandGroup is only initalized when Initalize() is run. I will run again with the command in the scheduler (which should run Initialize()) to see if there was a different exception that triggered originally, because I definitely tried with the command scheduler and also had a similar issue.

MrRedness, 2024-02-15:
Indeed it appears initialize is required to be run, and I was not running it.

closed 2024-02-15 by MrRedness

TheTripleV, 2024-02-15:
Commands are meant to be scheduled, not run manually by hand.