wpilibsuite/allwpilib #6383: [sysid] SysId crashing with wpilog file

bruingineer, 2024-02-20:
**Describe the bug**
SysId crashes when the "load" button is clicked when this log file and sysid tests are used.
I know the data does not look good. We need to re-run the tests, but I wanted to share the log file in case it is helpful to fix the crashing issue. The mechanism is a neo550 on a linear ball screw drive to change the angle of the shooter. the sensor is a rev through bore encoder measuring the angle of the system.

**To Reproduce**
Steps to reproduce the behavior:
1. load attached wpilog into sysid 2024.2.1 or 2024.3.1
2. Use the "simple" mechanism and "degrees" as units. I've also tested combinations of simple/meters, elevator/degrees, elevator/meters. All of them cause sysid to crash.
fields used are 
state: "sysid-test-state-ShooterElevation"
velocity: "Shuffleboard/subsytems/ShooterElevation/Speed", 
position: "Shuffleboard/subsytems/ShooterElevation/pos in Degrees",
voltage: "Shuffleboard/subsytems/ShooterElevation/Motor Voltage"

 - Link to log file:
[FRC_20240219_222307.zip](https://github.com/wpilibsuite/allwpilib/files/14352167/FRC_20240219_222307.zip)

**Desktop (please complete the following information):**
 - OS: windows 10
 - Project Information: 2024.2.1 or 2024.3.1

Anhysteretic, 2024-03-01:
Got the same issue here except that I am on mac m1. Here is my mac crash log if it may help debug the issue 
[sysid-2024-02-29-200828.ips.zip](https://github.com/wpilibsuite/allwpilib/files/14456348/sysid-2024-02-29-200828.ips.zip). Also just the .wpilog that I used:
[045E5F8B463847532020204B39470DFF_2024-02-29_18-49-21.hoot.wpilog.zip](https://github.com/wpilibsuite/allwpilib/files/14456349/045E5F8B463847532020204B39470DFF_2024-02-29_18-49-21.hoot.wpilog.zip).

binex-dsk, 2024-03-07:
Also facing this issue with arm/radians.

binex-dsk, 2024-03-07:
For our case we saw that some of Data was invalid, null, or NaN. Check to ensure your data isn't bad in any of those ways @Anhysteretic @bruingineer but this should not result in a crash (probably a simple check for invalid values)

calcmogul, 2024-03-13:
This crash no longer occurs as of #6425.

closed 2024-03-13 by calcmogul