wpilibsuite/allwpilib #6388: SysId position feedback gains do not account for measurement delay

brettle, 2024-02-21:
The following calls to `LatencyCompensate()` in `sysid::CalculatePositionFeedbackGains()` always pass 0 sec for the measurement delay. That makes the call a no-op. As a result, the position feedback gains are substantially higher than they should be when there is measurement delay (which there often is). Shouldn't these calls pass `preset.measurement_delay` instead of 0 sec?

https://github.com/wpilibsuite/allwpilib/blob/607682b68717e4632c1bba1f2e58bed203bd2032/sysid/src/main/native/cpp/analysis/FeedbackAnalysis.cpp#L35
https://github.com/wpilibsuite/allwpilib/blob/607682b68717e4632c1bba1f2e58bed203bd2032/sysid/src/main/native/cpp/analysis/FeedbackAnalysis.cpp#L50

calcmogul, 2024-02-21:
`preset.measurement_delay` is the velocity measurement delay, so it would be incorrect to use for position.

brettle, 2024-02-21:
Can't you use the value calculated for the "Position Measurement Delay (ms)" in the Feedforward Analysis pane? Fwiw, when I load data, that value also seems to get copied to the "Measurement Delay (ms)" value in the Feedback Analysis pane. At least in my particular case, they are identical.

brettle, 2024-02-21:
Actually, it looks like the following code sets `preset.measurement_delay` to the value calculated in the feedforward anaylsis for either the position measurement delay or the velocity measurement delay, depending on whether the user has selected "Position" or "Velocity" for the control loop type.

https://github.com/wpilibsuite/allwpilib/blob/607682b68717e4632c1bba1f2e58bed203bd2032/sysid/src/main/native/cpp/view/Analyzer.cpp#L686-L690

calcmogul, 2024-02-22:
Fair enough.