wpilibsuite/allwpilib #6407: DS Log Viewer visibility problem

chauser, 2024-03-02:
**Describe the bug**
The DS Log Viewer (24.0.2) does not display events (Events tab) at the very end of the log. They are past the bottom of the window. The scroll handle can actually be moved down to the point _it_ is no longer visible either. Very reminiscent of the problems in the graph view of the kickoff release of the game tools.

**To Reproduce**
Open .dslog and .dsevents. Switch to Event List tab. Observe that the end of the log is not visible.

**Expected behavior**
All events should be visible.

**Screenshots**
[image]
This is the farthest down you can see in this log file, but there are another 1+minutes worth of events. Note that the scroll handle is not visible. I had to move the mouse several inches below the bottom of the window to get this far down in the log.

**Desktop (please complete the following information):**
 - OS: Windows 11

spacey-sooty, 2024-03-03:
+1 I've seen this happen as well

sciencewhiz, 2024-11-30:
This appears fixed in the 2025 beta log viewer 25.0d5