wpilibsuite/allwpilib #6411: Command.isFinished() is misleading

GregoryLi360, 2024-03-05:
**Is your feature request related to a problem? Please describe.**
Command.isFinished() is misleading. Initially, you would expect it to be true whenever the command is not executing (namely, not between initialize() and end()), however, it can return false even after a command "finishes" and its end function is called. 

**Describe the solution you'd like**
Another method called "isRunning" or something similar that actually returns whether the command is or is not currently executing. Also better documentation of this method in commands like SequentialCommandGroup.

SamCarlberg, 2024-03-05:
It sounds like you're asking for `isFinished()` to behave like `isScheduled()`, which checks if a  command is currently scheduled to run. `isFinished()` is a required function for the command scheduler to use to detect when commands have completed their tasks and should be terminated; it's a non sequitur to ask a command that isn't running if it has completed.

GregoryLi360, 2024-03-05:
I agree, however, ```isScheduled()``` has some weird behaviors. For example, if a command is part of a command composition and not exposed to the scheduler, it will always return false.

Starlight220, 2024-03-05:
> if a command is part of a command composition and not exposed to the scheduler

In those cases, you shouldn't be using the object at all. C++ with CommandPtr has the correct semantics: the object is moved into the composition and is no longer accessible. Unfortunately, these semantics aren't possible in Java.

closed 2024-03-06 by Starlight220