wpilibsuite/allwpilib #6437: Multiple command compositions exception message truncated

sciencewhiz, 2024-03-13:
Running the following which composes the same command twice gives an unhelpful exception message "originally composed at:"

```
    Command testCommand = new InstantCommand();
    Command testCommand2 = testCommand.withTimeout(1);
    Command testCommand3 = testCommand.withTimeout(2);
```
The stack trace:
```
Error at frc.robot.Robot.autonomousInit(Robot.java:44): Unhandled exception: java.lang.Exception: Originally composed at:
        at edu.wpi.first.wpilibj2.command.CommandScheduler.registerComposedCommands(CommandScheduler.java:599)
        at edu.wpi.first.wpilibj2.command.ParallelRaceGroup.addCommands(ParallelRaceGroup.java:49)
        at edu.wpi.first.wpilibj2.command.ParallelRaceGroup.<init>(ParallelRaceGroup.java:35)
        at edu.wpi.first.wpilibj2.command.Command.raceWith(Command.java:310)
        at edu.wpi.first.wpilibj2.command.Command.withTimeout(Command.java:149)
        at frc.robot.Robot.autonomousInit(Robot.java:44)
        at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:345)
        at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:131)
        at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:365)
        at edu.wpi.first.wpilibj.RobotBase.lambda$startRobot$0(RobotBase.java:433)
        at java.base/java.lang.Thread.run(Thread.java:833)
```

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Project Information: 
```
WPILib Information:
Project Version: 2024.3.1
VS Code Version: 1.85.1
WPILib Extension Version: 2024.3.1
C++ Extension Version: 1.19.1
Java Extension Version: 1.26.2023121408
Java Debug Extension Version: 0.55.2023121302
Java Dependencies Extension Version 0.23.2023120100
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    WPILib-New-Commands (1.0.0)
```

Starlight220, 2024-03-13:
Oof. I guessed that this was happening.

The cause here is that RobotBase's global exception handler grabs the "cause" exception, stripping the one surrounding it.

Unless we change that, there's no way to report multiple stack traces.

Putting the original stack trace in the exception message is what is done in C++ (which I doubt works properly) but still feels quite clunky and unreadable - especially with our target demographic.

Searching the original stack trace for the first non-WPILib stack trace element and formatting it into the exception message could be done, but that would be a bit less helpful (especially with libraries such as PP that compose commands internally).

I don't have a good solution here, besides maybe changing the exception handler to not drop surrounding exceptions.

sciencewhiz, 2024-03-13:
We print multiple stack traces for multiple hardware allocations: https://docs.wpilib.org/en/stable/docs/software/basic-programming/reading-stacktraces.html#hal-resource-already-allocated

Starlight220, 2024-03-13:
Where is the code handling that?

sciencewhiz, 2024-03-14:
Looks like we do that in the HAL https://github.com/wpilibsuite/allwpilib/pull/3322

PR #6501 (2024-04-22): [cmd] Fix double composition error truncation issue
Fixes #6437

This is the resulting error message:
```
Error at frc.robot.MyRobot.robotInit(MyRobot.java:20): Unhandled exception: java.lang.IllegalArgumentException: Commands that have been composed may not be added to another composition or scheduled individually!
java.lang.Exception: Originally composed at:
        at edu.wpi.first.wpilibj2.command.Command.andThen(Command.java:252)
        at frc.robot.MyRobot.robotInit(MyRobot.java:19)
        at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:107)
        at edu.wpi.firs [... cut]
closed 2024-04-22 by PeterJohnson