wpilibsuite/allwpilib #6443: Sim GUI can't display protobuf data

Gold856, 2024-03-15:
**Describe the bug**
When publishing protobuf data, like a Pose2d or Translation2d, Sim GUI will display the fields (like `translation` and `rotation` for Pose2d or `x` and `y` for Translation2d) but it won't display the actual data.

**To Reproduce**
Steps to reproduce the behavior:
1. Use a ProtobufPublisher to publish an object like Pose2d.
2. Run the simulator.
3. Observe the lack of data, but the presence of the fields.

**Expected behavior**
Sim GUI should display the data like Glass does.

**Screenshots**
Glass showing data:
[image]

Sim GUI not showing data but showing fields:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - WPILib Information:
Project Version: 2024.3.2
VS Code Version: 1.84.0
WPILib Extension Version: 2024.3.1
C++ Extension Version: 1.19.1
Java Extension Version: 1.26.2023121408
Java Debug Extension Version: 0.55.2023121302
Java Dependencies Extension Version 0.23.2023120100
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    NavX (2024.1.0)
    CTRE-Phoenix (v6) (24.1.0)
    REVLib (2024.2.0)
    WPILib-New-Commands (1.0.0)

PeterJohnson, 2024-03-16:
Weird.  The same code is used for both.  Will look into it.

ThadHouse, 2024-08-18:
This is a bug exclusively on Windows, caused by invalid use on data exports across dll boundaries. Thats why this doesn't show in glass. Glass is entirely statically linked. #6979 when done right will fix it.

closed 2024-08-21 by PeterJohnson