wpilibsuite/allwpilib #6469: [discussion] Prefer LTV over Ramsete

DeltaDizzy, 2024-03-24:
Currently, the WPILib trajectory stack consists of `Trajectory` and related classes in addition to the following controllers and helper classes:
- `RamseteController`
- `RamseteCommand`
- `LTVUnicycleController`
- `LTVDifferentialDriveController`

`RamseteController` has opaque tuning handles, with those exposed by the LTV controllers being much more understandable at a glace (with the analogies of error tolerance and maximum control effort). Preferring LTV in examples and documentation could greatly improve the experiences of teams opting to do differential drive trajectory following. `LTVUnicycleController` would be presented as the "basic" option and (subject to discussion outcomes) a premade command type create for using it. `LTVDifferentialDriveController` would be presented as a more advanced and better UX option for teams who are comfortable with SysId and creating custom commands. frc-docs and examples changes would accompany this. Ramsete would not be deprecated but documentation would recommend using LTV instead as there are few if any advantages to it.

calcmogul, 2024-03-24:
We've actually discussed deleting all controller command classes, because they're more error-prone than just using the controller directly.

DeltaDizzy, 2024-03-24:
Which was met with much debate (#3350, #3366, #5352), so I guess consider this attempt 4 at a consensus? :p

Gold856, 2024-06-18:
Resolved by #6494.

closed 2024-06-18 by calcmogul