wpilibsuite/allwpilib #6478: DataLog should be able to record nulls

fovea1959, 2024-03-27:
**Is your feature request related to a problem? Please describe.**
We have situations where we want to log data which may contain null values: example: record the Pose2d that vision sees. It might be null if our camera is blocked.

**Describe the solution you'd like**
Change the signatures for the various DatalogEntry objects accept objects instead of primitives and record accordingly.

**Describe alternatives you've considered**
Team provided shims that deliver NaN or a zero length string (not tested). Doesn't work for Long, though. Don't know how downstream tools (AdvantageScope) deal with NaN, though.

**Additional context**
<none>

PeterJohnson, 2024-03-27:
The underlying data format has no way of storing null values.  It's perfectly valid to actually want to log a NaN, so we wouldn't want to arbitrary map null to NaN, and integers have no "invalid" values.

closed 2024-03-27 by PeterJohnson

jwbonner, 2024-03-27:
In regard to publishing vision poses, I would recommend logging it as an array of poses rather than a single value. AdvantageScope will correctly render empty arrays of objects.

fovea1959, 2024-03-27:
> In regard to publishing vision poses, I would recommend logging it as an array of poses rather than a single value. AdvantageScope will correctly render empty arrays of objects.

Thanks. Good workaround. Have to see how advantagescope renders it.

fovea1959, 2024-03-27:
> The underlying data format has no way of storing null values.  It's perfectly valid to actually want to log a NaN, so we wouldn't want to arbitrary map null to NaN, and integers have no "invalid" values.

Agreed. I was going to use that workaround in our team code, not suggesting it for wpilib.