wpilibsuite/allwpilib #6479: Unify the various DataLog*Entry to have a common superclass

fovea1959, 2024-03-27:
**Is your feature request related to a problem? Please describe.**
Had to write a lot of boilerplate to rework our legacy data logging to use DataLog. Having all the BooleanLogEntry, LongLogEntry, etc have a common ancestor using generics would have simplified the work.

**Describe the solution you'd like**
BooleanLogEntry, LongLogEntry, etc all are subclasses of a typable GenericLogEntry:

```
public class GenericLogEntry<T> {
  public void update (T value) {...}
  ...
}

public class BooleanLogEntry extends GenericLogEntry<Boolean> { ...
  // don't need to provide a type specific update method. <woot>!
}
```

**Describe alternatives you've considered**
Writing a lot of boilerplate.

**Additional context**
<none>

ThadHouse, 2024-03-27:
DataLog is designed to be fast, and generics here mean every value logged is an object allocation. That is the reason there isn't a base class.

Additionally, because we're having to pass the primitive values down to native code, the generic backing class would need to switch on the generic type, which would result in even slower logging.

fovea1959, 2024-03-27:
ok. I put another enhancement request in for being able to log null values, so I assume that's also a nogo for the same reason?

ThadHouse, 2024-03-27:
It has similar concerns, but other ones as well. Null isn't a valid value for primitives, so it would require changes to the binary format to support null values as well.

closed 2024-03-27 by PeterJohnson