wpilibsuite/allwpilib #6482: wpilog DS:autonomous signal not logged in some matches

willtoth, 2024-03-27:
Looking at a .wpilog, specifically the `DS:autonomous` signal, in advantage scope things look how I would expect them. The mode transitions from false to true and back:

[image]

However in some matches it looks like this:

[image]

This matches the exported data from the datalogtool in table:

[image]

Attached two data logs, E15 does not have any issue, E1 does https://drive.google.com/file/d/195NSlsXxMEwuHO2WLhTKbwRbiUMDErrE/view?usp=sharing

Starlight220, 2024-03-27:
Is it possible that the DS log fragmented in those matches? Without info about what happened in those matches, I'm not sure how much can be done.

willtoth, 2024-03-27:
What do you mean by fragmented? The entirety of the match is in the single wpilog (e.g. all timestamps throughout the match).