wpilibsuite/allwpilib #6484: Documentation for `PIDController.setIntegratorRange` is incorrect

chauser, 2024-03-27:
The in-code documentation for `PIDController.setIntegratorRange` says
```
/**
   * Sets the minimum and maximum values for the integrator.
   *
   * <p>When the cap is reached, the integrator value is added to the controller output rather than
   * the integrator value times the integral gain.
   *
   * @param minimumIntegral The minimum value of the integrator.
   * @param maximumIntegral The maximum value of the integrator.
   */
```
The described semantics are very puzzling, but the code doesn't do that -- it does what I'd expect.
Suggested replacement:
/**
   * Sets the minimum and maximum values for the integrator.
   *
   * <p>If the integrator value is outside the range [minimumIntegral..maximumIntegral]
   * the integral gain times the corresponding limit is added to the controller output
   * rather than the integral gain times the integrator value.
   *
   * @param minimumIntegral The minimum value of the integrator.
   * @param maximumIntegral The maximum value of the integrator.
   */

chauser, 2024-03-28:
The suggested rewording above is still wrong. 
```
/**
   * Sets the minimum and maximum values for the integrator.
   *
   * <p>If the integrator value is outside the range [minimumIntegral/ki..maximumIntegral/ki]
   * either minimumIntegral or maximumIntegral limit is added to the controller output
   * instead of the integral gain times the integrator value.
   * Additionally, the integrator value is set to minimumIntegral/ki or maximumIntegral/ki if 
   * the integrator value would fall outside the range [minimumIntegral/ki..maximumIntegral/ki].
   *
   * @param minimumIntegral The minimum contribution of the integral term
   * @param maximumIntegral The maximum contribution of the integral term
   * */
```
That is, `minimumIntegral` and `maximumIntegral` should be thought of as the minimum and maximum _contribution_ of the integral term to the final result and not as the minimum and maximum values of the integrator value. 

The above comment essentially recreates the code in English -- is there a better way to write the comment?

chauser, 2024-03-28:
Finally, there is no check that ensures `minimumIntegral <= maximumIntegral`. Should there be?

calcmogul, 2024-06-19:
Fixed by #6489.

closed 2024-06-19 by calcmogul