wpilibsuite/allwpilib #6498: C++ Command Based: Assertion failed: isHandleInSync() && “invalid iterator access!”

MrRedness, 2024-04-02:
Been battling a rare but persistent bug that shows up once every couple hours of driver practice (and also in 1 match at competition). The issue always seemingly occurs right around scheduling/cancelling commands dealing with our intake subsystem (although I haven’t been able to prove its origin). I can share that code if it helps. The closest I found to others encountering this issue was some commentary on a [PR last summer](https://github.com/wpilibsuite/allwpilib/pull/5470#issuecomment-1659500434) that mentioned wacky behavior when scheduling another command at the end of a command, which IS something we do, so if that’s undefined behavior we can try something else.

Hoping to see if anyone has thoughts: any ideas welcome and thanks!

spacey-sooty, 2024-04-02:
Could you share your code?

KangarooKoala, 2024-04-02:
+1 to sharing your code, especially all of the commands that schedule other commands and the commands that get scheduled by them. (If possible, a GitHub link or a .zip file of the entire repository would be preferable, so that we won't need to ask again for more snippets of code)

While I still want to look at the code to better understand the issue, you could also think about why you're scheduling another command in `End()` in the first place- Is there some functionality you can't get with `command1.AndThen(command2)`? (If you need to create the command at runtime, then you can use `frc::cmd::Defer()`, and if you run into requirements issues, you can look at [the WPILib docs](https://docs.wpilib.org/en/stable/docs/software/commandbased/command-compositions.html#scheduling-other-commands) to see if proxying is appropriate for your situation.)

MrRedness, 2024-04-02:
The reason for not using .AndThen() or proxy is that we wanted this command to be uncancellable, and so it shouldn't be linked to a WhileTrue() trigger that is held by the driver. This command should be run whenever a note has entered the intake to properly index it and so being cancelled would leave the intake in an undesirable state. We also ran into a lot of issues with proxying, although it is possible we just didn't understand how they work properly. We ended up just making all the intake subsystem commands not actually require the intake subsystem (I know this is pretty dumb but we messed with proxy commands for the better part of a week and made the judgement call to move on and ignore requirements; I don't see any reason why this would be related to the crashing issue).

Today, we removed the command scheduling another command and replacing it with a Trigger that runs upon a boolean being set at the end of the intakeCmd, but we still ran into the issue occasionally.

I'll upload a zip of our codebase here. It is quite large; we figure the issue lies somewhere in intaking logic, so be that IntakeSubsystem.cpp, the pivotAndIntakeCmd() in PivotCommands.cpp, or the faceAndIntakeNoteCmd() in RobotContainer.cpp. There are also some triggers for the intake in configureButtonBindings in RobotContainer.cpp.

MrRedness, 2024-04-02:
[src.zip](https://github.com/wpilibsuite/allwpilib/files/14842909/src.zip)

bathtubed, 2024-04-02:
My hypothesis after looking through WPILib and hearing the issue described is that the SmallPtrSet which holds the actively scheduled commands is growing during iteration, which is invalidating the iterator.

Most of the time no iterator is invalidated because they're within the "SmallSet" size of 12, but perhaps occasionally they're going from 12 commands to 13 and growing the set to using the std::set instead of the optimized vector, which is causing this iterator to be invalidated midloop.

https://github.com/wpilibsuite/allwpilib/blob/25ad6eafd5eeaf57ed25db2414a20838bf098399/wpilibNewCommands/src/main/native/cpp/frc2/command/CommandScheduler.cpp#L199-L226

I would think that scheduling commands at all within an End or an Execute would be undefined behavior based on this code, but it likely only manifests when going between 12 and 13 active commands.

You could try to build in release mode to disable the iterator checking. I imagine it would probably still do horrible things but maybe it'd have less a chance of crashing mid match.

MrRedness, 2024-04-02:
> I'd suggest discussing the issues you encountered (maybe in a separate issue?), since the ideal situation (that is, less bugprone) would be to have intake commands require the subsystem and understand the proxying semantics, but it's also understandable if you don't want to go down that route right now. (For example, maybe your next competition is really soon.)

Unfortunately, yeah we have our DCMP tomorrow. 

Thank you all for the input. However as I mentioned I've since moved away from scheduling the command as part of the FinallyDo() function and instead modified a Boolean that is linked to a trigger, and the issue continued.

MrRedness, 2024-04-02:
> Most of the time no iterator is invalidated because they're within the "SmallSet" size of 12, but perhaps occasionally they're going from 12 commands to 13 and growing the set to using the std::set instead of the optimized vector, which is causing this iterator to be invalidated midloop.

I believe when I checked the log we were at around 4-5 commands running

KangarooKoala, 2024-04-02:
`isHandleInSync()` && "invalid iterator access!" occurs in the code when a SmallPtrSet or DenseMap iterator is dereferenced or incremented. `isHandleInSync()` will be false when `incrementEpoch()` happens after the iterator is created, which happens in `SmallPtrSet::clear()` and `SmallPtrSet::insert()` and when a DenseMap grows or is cleared (`DenseMap::reserve()`, `DenseMap::clear()`, `DenseMap::try_emplace()`, `DenseMap::insert()`, `DenseMap::getOrInsertDefault()`, `DenseMap::FindAndConstruct()`, `DenseMap::operator[]`, `DenseMap::insert_as()`).
Maybe I'm missing something, but I didn't see any remaining cases in CommandScheduler.cpp where any of those methods are called while there are active iterators.
If possible, a stacktrace would be really helpful (though I don't remember how good C++ stacktrace support is, since my team uses Java), even if it isn't very deep.

MrRedness, 2024-04-02:
> Could you share more of the stacktrace? (Preferably the full stacktrace, though if it has to wait until you get robot access that's fine)

I can look for it, however that's about all that showed up in the log. (Exception thrown, assertion failed, and the line #). Is there a way to ask for a more full stack trace?

KangarooKoala, 2024-04-02:
What's the line number?

bathtubed, 2024-04-02:
@KangarooKoala ah you're right I was oversimplifying my analysis. I see that a Schedule during an Execute or End would go to toSchedule instead of the main scheduledCommands set.

MrRedness, 2024-04-02:
I unfortunately don't have the DS log only Rio log on the laptop at home, but this assertion is only checked in two places
[image]

MrRedness, 2024-04-02:
> What's the line number? Also, it looks like for C++ you'd need to attach a debugger, which might be tricky to do on the robot: https://docs.wpilib.org/en/stable/docs/software/basic-programming/reading-stacktraces.html

Yeah we've had terrible success attaching a debugger, and this issue happens rare enough that it would be difficult to attach every time. Also, I haven't been able to reproduce the issue in sim or even while repeatedly intaking. It seems to happen most often when we are running an intense, full practice match, so that tracks with the number of commands being high.

KangarooKoala, 2024-04-02:
Looking through your code (main/cpp/RobotContainer.cpp, line 1065), you run `CommandScheduler::Run()` and some other functions in a separate thread, and `CommandScheduler` isn't synchronized, so there's probably some operation that's done in a different thread that invalidates an iterator. (For example, if the transitions happen when starting and ending auto, then that could be caused by the auto command `Schedule()` calls in the `Robot` methods) Quickest fix might be to call `CommandScheduler::GetInstance().Run()` from `RobotPeriodic()`, so that all of the command scheduler stuff happens in the main robot thread.

MrRedness, 2024-04-03:
> Looking through your code (main/cpp/RobotContainer.cpp, line 1065), you run `CommandScheduler::Run()` and some other functions in a separate thread, and `CommandScheduler` isn't synchronized, so [...]

I had the same thought, however these issues never manifest in auto (which is the only place we schedule commands outside of the traditional trigger system, which should bind itself to the CommandScheduler.Run()). I don't see anywhere else that we could be scheduling/cancelling commands unless wpilib behind the scene is running some stuff on multiple threads.

Gold856, 2024-04-03:
Considering this happens rarely, and the fact that the command scheduler is run on a different thread, you are probably running into a race condition that causes the pointer iterator to be invalidated. The command scheduler isn't thread-safe, and I agree that you should try running the command scheduler synchronously with the main robot loop and see if the issues persist. I do see that you run the command scheduler faster than default, so you can use [AddPeriodic](https://docs.wpilib.org/en/latest/docs/software/convenience-features/scheduling-functions.html) to run it synchronously and with a lower period.

MrRedness, 2024-04-03:
>The command scheduler isn't thread-safe, and I agree that you should try running the command scheduler synchronously with the main robot loop and see if the issues persist.

Thanks for the advice. I've set it up and as long as we don't see performance impacts I guess there's no reason not to run synchronously.

bathtubed, 2024-04-03:
Personally, if I have a subsystem with specific state invariants I want to maintain, I control that outside of commands within the subsystem's Periodic() method.
Then I just configure commands to trigger these state transitions and wait for sensors to indicate completion of these transitions.

KangarooKoala, 2024-04-03:
Honestly I wouldn't be surprised if performance actually goes up a bit by cutting down on the number of threads, since the roboRIO only has 2 cores, so performance-wise, adding more threads past that point just adds overhead of switching between the threads.

Starlight220, 2024-04-04:
Unless I'm misreading things, the issue described here stemmed from multithreading and the scheduler not being thread-safe -- which is a non-goal.
Closing.

closed 2024-04-04 by Starlight220