wpilibsuite/allwpilib #6504: [math] Pose2d.minus has unexpected behavior; seems to rotate the translation

binex-dsk, 2024-04-08:
**Describe the bug**
We expect `Pose2d.minus(Pose2d other)` to return something along the lines of (pseudocode)

```
new Pose2d(translation - otherTranslation, rotation - otherTranslation);
```

However, it seems not only to go the wrong way (other - this), but seems to do some weird things where it rotates the pose  based on their rotation difference?

All in all, seems like this needs to be resolved via clearer documentation & possibly an alternate method that does what I expect it to do.

**To Reproduce**
Steps to reproduce the behavior:
1. Call `<Pose2d>::minus(Pose2d other)`

**Expected behavior**
A new Pose2d or Transform2d with the differences in the translational and rotational components of the poses, in the order called (this - other)

**Desktop (please complete the following information):**
 - OS: Artix Linux 6.8.1, Windows 10, Windows 11
 - Project Information:
 WPILib Information:
Project Version: 2024.3.2
VS Code Version: 1.85.1
WPILib Extension Version: 2024.3.2
C++ Extension Version: 1.19.1
Java Extension Version: 1.26.2023121408
Java Debug Extension Version: 0.55.2023121302
Java Dependencies Extension Version 0.23.2023120100
Java Version: 17
Java Location: /home/swurl/wpilib/2024/jdk
Vendor Libraries:
    PathplannerLib (2024.2.8)
    CTRE-Phoenix (v5) (5.33.1)
    CTRE-Phoenix (v6) (24.2.0)
    REVLib (2024.2.4)
    WPILib-New-Commands (1.0.0)
    photonlib (v2024.3.1)
    PlayingWithFusion (2024.03.09)

**Additional context**
Add any other context about the problem here.

calcmogul, 2024-04-08:
b.minus(a) creates a Transform2d c such that a.transformBy(c) should equal b. Is that not happening?

binex-dsk, 2024-04-08:
> b.minus(a) creates a Transform2d c such that a.transformBy(c) should equal b. Is that not happening?
> 
> ```java
> new Pose2d(translation - otherTranslation, rotation - otherTranslation);
> ```
> 
> That's not how Transform2d works.

Understood, and that's where the confusion comes from. It makes more sense to me that you should be able to subtract the translational and rotational components and have that as a separate method. Having `minus` be unrelated to that concept and instead into the realm of transformations is rather unintuitive (plus, relativeTo already exists, which minus calls anyway).

calcmogul, 2024-04-08:
It's not unrelated though. It's exactly what should happen to make the other operations, like linear interpolation, work correctly.

binex-dsk, 2024-04-08:
Okay, that makes much more sense now, since that's how vector subtraction/addition works. We were thinking about it differently...

However it caused a minor headache for us, since the naming of `minus` made us assume it was individual subtracting the vector components. Thinking about it now though, we never actually needed the rotational difference so we should've used `Translation2d.minus()` which gives us what we're looking for. Makes sense now that I understand we were operating on  `[x, y, t]`, not `[x, y]` and `[t]`. There *may* be room for improved documentation in case students aren't as familiar with  vector concepts.

With all that said, you *are* correct that it's not actually useful for anything now that I've thought about it, since separate operation on translation & rotation gives you the desired results.

closed 2024-04-08 by binex-dsk

calcmogul, 2024-04-08:
There may be room for improved documentation in case students aren't as familiar with vector concepts.

We already document the current behavior to be what I described: https://github.wpilib.org/allwpilib/docs/release/java/edu/wpi/first/math/geometry/Pose2d.html#minus(edu.wpi.first.math.geometry.Pose2d)

binex-dsk, 2024-04-08:
Yep that's what made me skeptical, however some of our students understood "relative to" as in `translation1 + translation2` and `rotation1 + rotation2`. There is room for improvement on our end but I do wonder if there is room for more easily accessible documentation on vectors (I think there exists something in frc-docs already) that is directly referenced in Transform2d

calcmogul, 2024-04-09:
Vector documentation doesn't seem relevant here though.