wpilibsuite/allwpilib #6513: Make LinearSystemId functions return LinearSystem with fully observable state

calcmogul, 2024-04-13:
LinearSystemId.IdentifyPositionSystem() returns a LinearSystem<2, 1, 1>, which is incompatible with the physics simulation classes.

Instead, it should return a LinearSystem<2, 1, 2>, and LinearSystem should have a function for slicing the states, inputs, and outputs so that it can produce a LinearSystem<2, 1, 1>.

narmstro2020, 2024-04-27:
I've wanted this fix for a bit myself.  Seemed like a weird gap after reading all of the documentation on the subject.  
I can take a whack at it.  This should also apply to createElevatorSystem and createSingleJointedArmSystem.  Flywheel, DCMotor, and the DriveTrain should be left alone if I'm not mistaken?

calcmogul, 2024-04-27:
Yea, those three already do the correct thing.

narmstro2020, 2024-04-27:
On a marginally related subject I'd also like to take the type to add a WheelSim class that simulates something like a drive wheel.  It would effectively be the same as DCMotorSim, but reading out linear units instead of angular ones.

calcmogul, 2024-04-27:
When would that be used? Seems like a lot of API overhead for a single scalar multiplication on the user side.

narmstro2020, 2024-04-27:
My use case would be in terms of simulating a drive wheel given starting with kV and kA values in VoltsPerMPS instead of RadPerSec.  

Perhaps a constructor that takes in kV and kA in the linear units along with a conversion factor in the existing DCMotorSim class?

calcmogul, 2024-04-27:
That's a non-starter in C++ because it isn't unit-safe.

narmstro2020, 2024-04-27:
Okay.  I'm content without.  Just an idea.

calcmogul, 2024-09-13:
Fixed by #6554.