wpilibsuite/allwpilib #6532: scheduling from Command.end?

truher, 2024-04-23:
in Command.end(), theres a comment that advises against using this method to schedule another command, suggesting andThen(), which I think would only work for a static sequence.  I had in mind to use the end() method to do ad hoc command sequencing, and it seems like it should work fine.  is there a reason not to?

spacey-sooty, 2024-04-23:
What is scheduling from `end` going to gain you compared to using factory methods? You might be looking for `select` or `either`

truher, 2024-04-23:
maybe?  how do I use select() or either() to assemble an indefinite series of arbitrary commands?

spacey-sooty, 2024-04-23:
Something like `.run(select(...))` could work?

truher, 2024-04-23:
i see, you mean to put the selection logic in the command selector.  how do you trap the command completion to select the next one?  I think my main question is actually the reasoning behind the comment.  why say that chaining commands through end() is a bad idea if it seems to work fine?

spacey-sooty, 2024-04-23:
I presume that it isn't recommended as it adds a side effect to the command that is essentially hidden. When you use a factory command you explicitly state that after this command ends this command should run. Compared to end where you are implicitly moving on to it which could cause unintuitive behaviour using something like factory methods as well.

truher, 2024-04-23:
ok thanks.  maybe this is more of a question about canon rather than the specific Command.end idea.  I'll ask over on CD.

KangarooKoala, 2024-04-23:
Looks like that line was originally added in #2450 (to resolve #2441). However, the behavior that needed to be fixed (scheduling a command with shared requirements from `end()` would cause the original command to be canceled, calling `end()` again in an infinite loop) has been fixed in #5470, so we probably don't need that comment anymore.

Regarding https://www.chiefdelphi.com/t/canonical-command-chaining/464156 (which I'm guessing is where you asked on CD), overriding `Command.end()` (or equivalent, wrapping in `onEnd()`) to get a callback when the command finishes should work, though if you run into bugs a) please file a bug report to help us improve! and b) try polling [`Command.isScheduled()`](https://github.wpilib.org/allwpilib/docs/release/java/edu/wpi/first/wpilibj2/command/Command.html#isScheduled()) periodically to detect when it is no longer scheduled.
Depending on your framework, you also might be able to use `cmd.andThen(Commands.defer(() -> getNextCommand(), Set.of(<requirements>)))` instead of wrapping in `cmd.onEnd(() -> scheduleNextCommand())` or overriding `cmd.end()` to schedule the next command.

closed 2024-06-19 by calcmogul