wpilibsuite/allwpilib #6538: New Name For Command Decorator .deadlineWith()

tom131313, 2024-04-25:
The command decorator `.deadlineWith()` implies the command being decorated is of an equal level to the commands that follow within the parentheses. That causes confusion as to what is the deadline.

I suggest a clearer, more descriptive name is `.deadlineFor()` with the better implication that the command is the deadline and the following commands within the parentheses are the commands that will die with the completion of the first command that is decorated.

Obviously this is a breaking change so both decorator names would have to be maintained for some period of time.

Starlight220, 2024-04-26:
Sounds like a good idea.

Classic deprecation cycle:
2025: add new name, deprecate old
2026: remove old

closed 2024-04-28 by PeterJohnson

sciencewhiz, 2024-04-28:
This needs an frc-docs PR