wpilibsuite/allwpilib #6586: be able to use InterpolatingDoubleTreeMap.put().put()... inline

ori-coval, 2024-05-05:
**Is your feature request related to a problem? Please describe.**
for me it is a bit annoying not being able to use .put when creating a new tree map and also not being able to use .put() in a constants class. 

**Describe the solution you'd like**
solution: return the tree when using .put() so you can do as the example below and add all of the data inline.
        `public static final InterpolatingDoubleTreeMap INTERPOLATION_MAP = new InterpolatingDoubleTreeMap().put(2.9, 62.0).put(2.52 , 55.0).put(2.15 , 50.0).put(3.59, 63.0).put(3.04, 56.0);`

ori-coval, 2024-05-05:
if this is decided to be done I have no prob implementing it.

spacey-sooty, 2024-05-05:
I think so long as you don't have to allocate a new object each time this is a fine feature to add

ori-coval, 2024-05-05:
no new allocation just return this

SamCarlberg, 2024-05-05:
An API shape like Java's existing `Map.of` but accepting a variadic `double` array would be preferred.

```java
InterpolatingDoubleTreeMap interpolationMap =
  InterpolatingDoubleTreeMap.of(
    2.90, 62,
    2.52, 55,
    2.25, 50,
    3.59, 63,
    3.04, 56
  );
```

ori-coval, 2024-05-05:
I can see why some people would prefer a variadic double array but I can say for myself that I prefer chaining the .put() method for a clear separation between each pair.

so how about adding both?

PeterJohnson, 2024-05-05:
I assume the put() call was modeled after the Java Map put() function, which returns the previous value, not the map itself.  https://docs.oracle.com/javase/8/docs/api/java/util/Map.html#put-K-V-

ori-coval, 2024-05-05:
currently put() returns void

ori-coval, 2024-05-05:
the reason is ease of use and readability.  

I prefer the lower example it makes it easier and faster to add multiple values one after the other and keeps a clear separation between the added values, unlike a variadic double array. 
a variadic double array would also be good enough for me but I think chaining would be easier, especially for beginners.
```
    static {
        shooterAngleInterpolation.put(2.9, 62.0);
        shooterAngleInterpolation.put(2.52, 55.0);
        shooterAngleInterpolation.put(2.15, 50.0);
        shooterAngleInterpolation.put(3.59, 63.0);
        shooterAngleInterpolation.put(3.04, 56.0);
    }
```

```

        shooterAngleInterpolation.put(2.9, 62.0)
        .put(2.52, 55.0)
        .put(2.15, 50.0)
        .put(3.59, 63.0)
        .put(3.04, 56.0);}
```

KangarooKoala, 2024-05-05:
What if instead of `put()` it was named `addEntry()`, `withEntry()`, or something else which doesn't have an existing meaning/convention like `put()`?

calcmogul, 2024-05-06:
We could implememt this function from the standard library:
https://docs.oracle.com/en/java/javase/17/docs/api/java.base/java/util/Map.html#ofEntries(java.util.Map.Entry...)

ori-coval, 2024-05-08:
okay, no prob.
just making sure this should be added to InterpolatingTreeMap so it would affect all types not just InterpolatingDoubleTreeMap right?

PR #6635 (2024-05-20): [wpimath] Add InterpolatingDoubleTreeMap.of()
Resolves #6586
closed 2024-05-20 by PeterJohnson