wpilibsuite/allwpilib #6610: Add abseil-cpp as a 3rd party library

pjreiniger, 2024-05-11:
I wanted to upgrade protobuf as the more recent versions have better bazel support. The currently vendor'd version is from 9/22. Moving to a more recent release requires a dependency on `abseil-cpp`. I saw the #6607  and #6609, which also refer to abseil, albeit in a cmake case.

I'm happy to do the work, I just wanted approval for such a change before spending time on it.

PeterJohnson, 2024-05-11:
Other than the fact I hate abseil with the flame of 1000 suns because it is the definition of dependency hell, if we have to upgrade, we have to upgrade.  Can we at least build it all into a single library rather than the 50+ individual libraries it seems to have in distributions?

Gold856, 2024-05-27:
I think reviving the [thirdparty-protobuf repo](https://github.com/wpilibsuite/thirdparty-protobuf), optionally applying some patches, and building it like we do with OpenCV (Gradle invokes CMake, copies the libraries, and bundles it into a zip file artifact) is the way to go. CMake will stuff the 50+ libraries into one folder we can bundle up.

PR #7988 (2025-08-17): Replace libprotobuf with upb for dynamic decode
libprotobuf is a very annoying dependency to deal with, and with the switch to nanopb for generated C++ code, libprotobuf is only used for dynamic decode in the GUI apps. libprotobuf has been swapped out with upb, a much smaller C-based library that supports reflection and can therefore do dynamic decode. This means we can remove the libprotobuf dependency and stop dealing with build issues because of it. Closes #6610 and closes #7250.
closed 2025-08-17 by PeterJohnson