wpilibsuite/allwpilib #6622: Deprecate robotInit()

calcmogul, 2024-05-13:
The constructor is the language-level equivalent, and they run at the same time at startup.

spacey-sooty, 2024-05-14:
Do we wanna deprecate this removing its behaviour immediately or leave the behaviour then remove it when we remove it?

Starlight220, 2024-05-14:
Deprecated functionality should still work until it is removed. If functionality is broken, it should be outright removed and not just deprecated.

KangarooKoala, 2024-07-27:
Was this issue completely resolved? The linked PR was changed to only update examples.

spacey-sooty, 2024-07-28:
> Was this issue completely resolved? The linked PR was changed to only update examples.

I opened https://github.com/wpilibsuite/allwpilib/issues/6845 to track this after that was merged. One of them should be closed as a duplicate.

PR #6623 (2024-07-17): Prepare for RobotInit deprecation by updating examples
Resolves #6622
PR #8199 (2025-09-09): [wpilib] Remove robotInit()
Fixes #6622.
commit be72d543 (2025-09-09): [wpilib] Remove robotInit() (#8199)

Fixes #6622.