wpilibsuite/allwpilib #6680: Make DriverStation error and warning reporting easier in C++

calcmogul, 2024-05-31:
**Is your feature request related to a problem? Please describe.**
In C++, we currently have an `FRC_ReportError()` macro and `frc::ReportError()` function in `frc/Errors.h`. These are very hard to find compared to Java's `DriverStation.reportError()` and `DriverStation.reportWarning()` functions. It's also awkward and unintuitive how you're supposed to reports warnings by calling `FRC_ReportError(warn::Warning, ...)`.

**Describe the solution you'd like**
I'd like ReportError() and ReportWarning() functions added to the C++ DriverStation class like Java has.

spacey-sooty, 2024-05-31:
I feel like doing `DriverStation::ReportWarning()` and `DriverStation::ReportError()` as wrappers of `FRC_ReportError(warn::Warning, ...)` and adding a `warn::Error` then wrapping `FRC_ReportError(warn::Error, ...)` would be a good solution

KangarooKoala, 2024-05-31:
Note, however, that the macro is able to automatically get the filename, line number, and function name for the user. Adding a `FRC_ReportWarning()` macro would definitely be good, though.

PeterJohnson, 2024-05-31:
I think the best thing we can do for now is add a FRC_ReportWarning().