wpilibsuite/allwpilib #6726: Question about New Radio from Vivid Hosting wireless connection

prishadargan, 2024-06-09:
For the new radios for FRC from Vivid Hosting, when trying to connect to the Team Access Point Mode, the radio is configured however the wifi won't appear when trying to connect it to a WiFi 6E. The radio is up to date with the latest firmware.

From the website:
When configured as a WIRELESS ACCESS POINT, due to regulatory requirements, the VH-109 radio cannot be powered using a battery source and is for Indoor Use Only. The device is regulated by the FCC LPI (Low Power Indoor) requirements when in this mode. 
Only when the VH-109 radio is configured as an INDOOR CLIENT DEVICE (STA) may it be powered using a battery.

I am unsure if this is related to the issue but at the moment our wireless connection does not work, and I am struggling to figure out the source of the problem.

PeterJohnson, 2024-06-09:
How is the radio powered?  It must be powered from a wall power converter, (not battery powered on the robot) to use it as a 6 GHz AP.  You can use the 2.4 GHz AP mode on a robot, or use two radios (one fixed, one on the robot) in 6 GHz mode.

prishadargan, 2024-06-09:
[image]

When we have it in radio mode and still the wifi connection option is not showing up.

PeterJohnson, 2024-06-09:
Is the physical DIP switch to the enable 2.4 GHz AP turned on?  I think it’s switch number 3.

prishadargan, 2024-06-09:
This worked! Thank you so much!

closed 2024-06-09 by prishadargan