wpilibsuite/allwpilib #6737: Is there a reason that closing a NetworkTables publisher doesn't cause it most recent data to be flushed?

brettle, 2024-06-12:
**Describe the question you have.**
Consider the following code:

```Java
try (var publisher = topic.publish()) {
    publisher.set(value);
}
```
Shouldn't remote subscribers (eventually) see the new value? My testing seems to indicate that they might not see it unless `inst.flushLocal()` (or `inst.flush()`) is called before the publisher is (auto-)closed. That seems to hold independent of any `PubSubOption`s associated with the publisher or whether the topic is cached . Is that intentional?

**Describe the reason for your confusion.**
The current behavior was just not what I expected. If it is intentional, it should probably be better documented.

PeterJohnson, 2024-06-12:
It used to, but I think the transmit changes made in #5659 are likely what causes these to be lost now.

brettle, 2024-06-12:
Is this considered a bug or was it considered a necessary part of #5659?

PeterJohnson, 2024-06-13:
I consider it to be a bug.