wpilibsuite/allwpilib #6762: Add 2D addressable LED display class

PeterJohnson, 2024-06-21:
Addressable LED matrix (2D) displays come in various display configurations and orientations (for example, row vs column major, serpentine, etc).  The simulation GUI [has support](https://github.com/wpilibsuite/allwpilib/blob/main/glass/src/lib/native/include/glass/support/ExtraGuiWidgets.h#L61) for these configurations, but there is no corresponding user code class to make it easy to use these displays as if they were a "standard" pixel array (e.g. row major, non-serpentine).

Braykoff, 2024-07-19:
Are you thinking like a LEDMatrixBuffer that extends LEDBuffer, or a separate class? Should there be LEDMatrixPattern support? Also, how could this be implemented in c++, which doesn’t currently have any matrix buffer support yet?

PeterJohnson, 2024-07-20:
I was envisioning a separate class.  In C++, just an OpenCV style Mat with pixel setting functions would do fine I think?