wpilibsuite/allwpilib #6785: cmake build broken as of 2024.1.1

cgpadwick, 2024-06-29:
**Describe the bug**
When using cmake to build the wpilib code on version 2024.1.1, encounter the following error:

```
make[2]: Leaving directory '/home/cpadwick/code/allwpilib/build'
[ 22%] Built target ntcore_test
make  -f wpimath/CMakeFiles/wpimath.dir/build.make wpimath/CMakeFiles/wpimath.dir/depend
make[2]: Entering directory '/home/cpadwick/code/allwpilib/build'
wpimath/CMakeFiles/wpimath.dir/build.make:73: *** target pattern contains no '%'.  Stop.
make[2]: Leaving directory '/home/cpadwick/code/allwpilib/build'
make[1]: *** [CMakeFiles/Makefile2:1250: wpimath/CMakeFiles/wpimath.dir/all] Error 2
make[1]: Leaving directory '/home/cpadwick/code/allwpilib/build'
make: *** [Makefile:166: all] Error 2
```

If you back up to version 2023.4.3 the compile proceeds without issue.

**To Reproduce**
Steps to reproduce the behavior:
1. from a linux machine, git clone the repo, e.g. `mkdir ~/code; cd code; git clone https://github.com/wpilibsuite/allwpilib.git`
2. checkout the 2024.1.1 tag, or any 2024.x tag: `cd allwpilib; git checkout tag/2024.1.1`
3. install some deps like opencv-dev
4. configured cmake as follows: `cd allwpilig; mkdir build; cmake -B build -DJAVA_OFF; cd build; cmake --build . --target wpimath` 
5. observe error

 - Link to code:
 - https://github.com/wpilibsuite/allwpilib.git

**Expected behavior**
I expected version 2024.x would build seamlessly with cmake.

**Screenshots**
No screenshot, error message posted above

**Desktop (please complete the following information):**
 - OS: linux mint, Release 21.3, Codename Virginia

**Additional context**
I tried other versions of 2024.x and had the same issue.  I backed up to 2023.4.3 and it worked fine.

PeterJohnson, 2024-06-29:
Try using the ninja generator instead.

cgpadwick, 2024-06-29:
ah ok.  Mind if I submit a PR on README-CMAKE.md to clarify that?

PeterJohnson, 2024-06-29:
That would be great, thanks!

cgpadwick, 2024-06-29:
hmmm, I still seem to be having an issue:

```
cpadwick@photonvision:~/code/allwpilib$ cmake -B build -G "Ninja" -DWITH_JAVA=OFF
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Platform version: 
-- Found LIBSSH: /usr/include (found suitable version "0.9.6", minimum required is "0.7.1") 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found version "3.12.4") 
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Found Threads: TRUE  
-- Found libatomic: /usr/lib/x86_64-linux-gnu/libatomic.so.1
-- Found libatomic: /usr/lib/x86_64-linux-gnu/libatomic.so.1
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Including X11 support
-- Found X11: /usr/include   
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter Development Development.Module Development.Embed 
Downloading /home/cpadwick/code/allwpilib/build/_deps/gl3w-build/include/GL/glcorearb.h...
Downloading /home/cpadwick/code/allwpilib/build/_deps/gl3w-build/include/KHR/khrplatform.h...
Parsing glcorearb.h header...
Generating /home/cpadwick/code/allwpilib/build/_deps/gl3w-build/include/GL/gl3w.h...
Generating /home/cpadwick/code/allwpilib/build/_deps/gl3w-build/src/gl3w.c...
-- Found OpenCV: /usr/local (found version "4.9.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cpadwick/code/allwpilib/build

cpadwick@photonvision:~/code/allwpilib$ cmake --build build
ninja: error: 'wpimath/protobuf::protoc', needed by 'wpimath/protobuf/controller.pb.cc', missing and no known rule to make it

```

Gold856, 2024-06-29:
Yeah, I wasn't reproduce your issue, and this seems to confirm it's a protobuf issue. Try running ccmake, toggle on advanced mode, go to the protobuf variables, and see if CMake has found a protoc executable. If it hasn't found a protoc executable, enter the path to a protoc executable, and try rebuilding.

cgpadwick, 2024-06-29:
Thanks for the info @Gold856  - that was indeed the issue. Also thanks for the tip on ccmake - I'd never seen that before and it is very useful! For future people who run into this issue, the fix on debian based systems was `sudo apt install protobuf-compiler`

closing

PeterJohnson, 2024-06-29:
It would be better if the cmake build detected the lack of a protobuf compiler and failed during configuration stage.

cgpadwick, 2024-06-30:
@PeterJohnson here is a patch.  I can't push to the repo otherwise I'd open a PR.

```
cpadwick@photonvision:~/code/allwpilib$ git diff main
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 095165276..b123d12d2 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -207,6 +207,7 @@ find_package(LIBSSH 0.7.1)
 set(CMAKE_FIND_PACKAGE_PREFER_CONFIG ON)
 set(protobuf_MODULE_COMPATIBLE ON CACHE BOOL "" FORCE)
 find_package(Protobuf REQUIRED)
+find_program(PROTOC_COMPILER protoc REQUIRED)
 set(CMAKE_FIND_PACKAGE_PREFER_CONFIG OFF)
 
 get_property(isMultiConfig GLOBAL PROPERTY GENERATOR_IS_MULTI_CONFIG)
```

PeterJohnson, 2024-06-30:
Typically the way to open a PR on GitHub is to fork the repo to your personal account, create a branch there, and then open a PR across repos.

cgpadwick, 2024-06-30:
ah got it, thanks!  just submitted https://github.com/wpilibsuite/allwpilib/pull/6792

Starlight220, 2024-07-01:
Closing as fixed by #6792

closed 2024-07-01 by Starlight220