wpilibsuite/allwpilib #6826: #6767 drastically increases test times

pjreiniger, 2024-07-12:
I don't know if this would be a problem in the real world but the change #6767 causes the wpimath tests to take a non-trivial amount of extra time. 

CMake - [Before](https://github.com/wpilibsuite/allwpilib/actions/runs/984, 1021325/job/27166656196) 0.35 sec, [After](https://github.com/wpilibsuite/allwpilib/actions/runs/9627657753/job/26555012698)  3.07 sec -> ~10x

Gradle - [Before](https://github.com/wpilibsuite/allwpilib/actions/runs/9841021311/job/27166657307) - 364ms, [After](https://github.com/wpilibsuite/allwpilib/actions/runs/9627657756/job/26555012370) 43715ms - ~120x (!!)

The really bad offending tests are when a invalid spline is getting testing ex `QuinticHermiteSplineTest.ThrowsOnMalformed`. If a team makes a bad spline live they will get plenty of warnings but their code might lock up.

PeterJohnson, 2024-07-12:
That sounds like a bug in the spline code, as it shouldn't be creating (0,0) rotations?

pjreiniger, 2024-07-12:
I didn't look into it, not since they are all named "throwOnMalformed" it might be an intended garbage in garbage (and exception) out

PR #6827 (2024-07-13): [wpimath] Exit early when parameterizing malformed spline
Currently, a max iteration heuristic is used to determine when a spline is malformed. Instead, we can report a failure immediately if dx and dy are too small, because the heading won't be accurate either.

Fixes #6826.
commit deb5f3d7 (2024-07-13): [wpimath] Exit early when parameterizing malformed spline (#6827)

Currently, a max iteration heuristic is used to determine when a spline
is malformed. Instead, we can report a failure immediately if dx and dy
are too small, because the heading won't be accurate either.

Fixes #6826.
closed 2024-07-13 by PeterJohnson