wpilibsuite/allwpilib #6837: "Can't load AMD 64-bit .dll on a IA 32-bit platform"

rnadu, 2024-07-14:
**Describe the bug**
Error when trying to run robot simulation.

**To Reproduce**
Steps to reproduce the behavior:
1. Installed according to directions [here](https://docs.wpilib.org/en/latest/docs/zero-to-robot/step-2/wpilib-setup.html)
2. Pressed the "Open WPIlib command palette" in vscode
3. Ran simulate robot code
4. Ran Sim GUI

**Expected behavior**
Runs simulation

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Project Information: 

```
WPILib Information:
Project Version: 2024.3.2
VS Code Version: 1.85.1
WPILib Extension Version: 2024.3.2
C++ Extension Version: 1.19.1
Java Extension Version: 1.26.2023121408
Java Debug Extension Version: 0.55.2023121302
Java Dependencies Extension Version 0.23.2023120100
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    NavX (2024.0.1-beta-4)
    CTRE-Phoenix (v6) (24.0.0-beta-8)
    REVLib (2024.2.0)

```

**Additional context**

This is the error log:
```
PS C:\Users\Riya\Training-StateMachines>  ${env:HALSIM_EXTENSIONS}='C:\Users\Riya\Training-StateMachines\build\jni\release\halsim_gui.dll;'; ${env:PATH}='C:\Users\Riya\Training-StateMachines\build\jni\release;C:\WINDOWS\system32\'; & 'C:\Program Files (x86)\Eclipse Adoptium\jdk-11.0.21.9-hotspot\bin\java.exe' '-agentlib:jdwp=transport=dt_socket,server=n,suspend=y,address=localhost:55766' '@C:\Users\Riya\AppData\Local\Temp\cp_8gb2v8ir6ui4wf02v1lx7b3uf.argfile' 'frc.robot.Main'
C:\Program Files (x86)\Eclipse Adoptium\jdk-11.0.21.9-hotspot\bin\java.exe
java.io.IOException: wpiHaljni could not be loaded from path or an embedded resource.
        attempted to load for platform /windows/x86/
C:\Users\Riya\Training-StateMachines\build\jni\release\wpiHaljni.dll: Can't load AMD 64-bit .dll on a IA 32-bitA common cause of this error is missing the C++ runtime.
Download the latest at https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads

        at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:98)
        at edu.wpi.first.hal.JNIWrapper.<clinit>(JNIWrapper.java:48)
        at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:415)
        at frc.robot.Main.main(Main.java:30)
```

spacey-sooty, 2024-07-14:
You appear to be on 32 windows which is unsupported

PeterJohnson, 2024-07-14:
It might be 64-bit windows, but you’re running a 32-bit JDK (Eclipse Adoptium 11) that is not the one WPILib installs.  It shouldn’t be picking that up is you are running the WPILib install of vscode.

sciencewhiz, 2024-07-15:
What are the contents of `C:\Users\Public\wpilib\2024\vscode\data\user-data\User\settings.json`?

rnadu, 2024-07-16:
I'm definitely on 64 bit windows. 

Contents of my C:\Users\Public\wpilib\2024\vscode\data\user-data\User\settings.json:
```
{
  "java.jdt.ls.java.home": "C:\\Users\\Public\\wpilib\\2024\\jdk",
  "extensions.autoUpdate": false,
  "extensions.autoCheckUpdates": false,
  "extensions.ignoreRecommendations": true,
  "update.mode": "none",
  "update.showReleaseNotes": false,
  "java.completion.matchCase": "off",
  "terminal.integrated.env.windows": {
    "JAVA_HOME": "C:\\Users\\Public\\wpilib\\2024\\jdk",
    "PATH": "C:\\Users\\Public\\wpilib\\2024\\jdk\\bin;${env:PATH}"
  }
}
```
FWIW, `C:\Users\Public\wpilib\2024\jdk\bin\java.exe` definitely exists.
```
PS C:\Users\Riya> C:\Users\Public\wpilib\2024\jdk\bin\java.exe --version
openjdk 17.0.8.1 2023-08-24
OpenJDK Runtime Environment Temurin-17.0.8.1+1 (build 17.0.8.1+1)
OpenJDK 64-Bit Server VM Temurin-17.0.8.1+1 (build 17.0.8.1+1, mixed mode, sharing)
```
Any other diagnostics that would help?
Thank you for the help!

sciencewhiz, 2024-08-07:
There's a few strange things happening. 

> S C:\Users\Riya\Training-StateMachines>  ${env:HALSIM_EXTENSIONS}='C:\Users\Riya\Training-StateMachines\build\jni\release\halsim_gui.dll;'; [...]
C:\Program Files (x86)\Eclipse Adoptium\jdk-11.0.21.9-hotspot\bin\java.exe

This is picking up the bolded JDK instead of the wpilib JDK. The settings.json looks ok, so it must be set somewhere else. The code for detecting the JDK looks for the setting in settings.json first. https://github.com/wpilibsuite/vscode-wpilib/blob/cb574f6cf4df7e25089362a8d3c87d15a21d9e01/vscode-wpilib/src/jdkdetector.ts#L40-L123

That code has a bunch of logging. Can you post `C:\Users\Public\wpilib\2024\logs\wpiliblog.txt`

rnadu, 2024-08-07:
Added the log file below. Doesn't seem helpful, see the comments before opening.
[wpiliblog.txt](https://github.com/user-attachments/files/16519467/wpiliblog.txt)

Added the following log lines, and got the following log:

```
PS C:\Users\Riya\Training-StateMachines> git diff HEAD
--- a/gradlew.bat
+++ b/gradlew.bat
@@ -69,7 +69,12 @@ goto fail
 @rem Setup the command line
+echo "CLASS PATH:"^M
+echo "%CLASSPATH%"^M
 
+echo "JAVA VERSION:"^M
+echo "%JAVA_EXE%"^M
+echo "%JAVA_EXE%" %DEFAULT_JVM_OPTS% %JAVA_OPTS% %GRADLE_OPTS% "-Dorg.gradle.appname=%APP_BASE_NAME%" -classpath "%CLASSPATH%" org.gradle.wrapper.GradleWrapperMain %*^M

```

```
"CLASS PATH:"
"C:\Users\Riya\Training-StateMachines\\gradle\wrapper\gradle-wrapper.jar"
"JAVA VERSION:"
"C:\Users\Public\wpilib\2024\jdk/bin/java.exe"
"C:\Users\Public\wpilib\2024\jdk/bin/java.exe" "-Xmx64m" "-Xms64m"   "-Dorg.gradle.appname=gradlew" -classpath "C:\Users\Riya\Training-StateMachines\\gradle\wrapper\gradle-wrapper.jar" org.gradle.wrapper.GradleWrapperMain simulateExternalJavaRelease   -Dorg.gradle.java.home="C:\Users\Public\wpilib\2024\jdk"
```

This is before clicking "Simulate Robot Code". So looks like everything before calling into gradle is fine, but gradle somehow picks up a bad java version.

sciencewhiz, 2024-08-08:
The log doesn't show the bad jdk being injected into the path, so that's happening later.

What are all the VS Code extensions you have installed?
Can you post the project .vscode\settings.json?

rnadu, 2024-08-08:
```{
  "java.configuration.updateBuildConfiguration": "automatic",
  "java.server.launchMode": "Standard",
  "files.exclude": {
    "**/.git": true,
    "**/.svn": true,
    "**/.hg": true,
    "**/CVS": true,
    "**/.DS_Store": true,
    "bin/": true,
    "**/.classpath": true,
    "**/.project": true,
    "**/.settings": true,
    "**/.factorypath": true,
    "**/*~": true
  },
  "wpilib.selectDefaultSimulateExtension": true,
  "java.test.config": [
    {
      "name": "WPIlibUnitTests",
      "workingDirectory": "${workspaceFolder}/build/jni/release",
      "vmargs": [ "-Djava.library.path=${workspaceFolder}/build/jni/release" ],
      "env": {
        "LD_LIBRARY_PATH": "${workspaceFolder}/build/jni/release" ,
        "DYLD_LIBRARY_PATH": "${workspaceFolder}/build/jni/release"
      }
    },
  ],
  "java.test.defaultConfig": "WPIlibUnitTests",
  "java.checkstyle.configuration": "${workspaceFolder}/checks.xml"
}
```
Ok, here it is. Thank you for helping.

sciencewhiz, 2024-08-08:
> What are all the VS Code extensions you have installed?

I didn't see an answer to this. My theory is you have another extension that is injecting the bad JDK.

> `"java.checkstyle.configuration": "${workspaceFolder}/checks.xml"`

This would imply you have the checkstyle extension installed.

rnadu, 2024-08-09:
[image]
These are all the extensions I have installed. They're all the default ones installed. The program doesn't work if any of the Java ones are disabled.

sciencewhiz, 2024-08-09:
Can you post the java extension log file? `ctrl-shift-p` and select `Java: Open Java Extension Log file`

rnadu, 2024-08-10:
```{
  message: "Use the JDK from 'java.jdt.ls.java.home' setting as the initial default project JDK.",
  level: 'info',
  timestamp: '2024-08-09 22:46:41.249'
}
{
  message: 'Starting Java server with: C:\\Users\\Public\\wpilib\\2024\\jdk\\bin\\java --add-modules=ALL-SYSTEM --add-opens java.base/java.util=ALL-UNNAMED --add-opens java.base/java.lang=ALL-UNNAMED --add-opens java.base/sun.nio.fs=ALL-UNNAMED -Declipse.application=org.eclipse.jdt.ls.core.id1 -Dosgi.bundles.defaultStartLevel=4 -Declipse.product=org.eclipse.jdt.ls.core.product -Djava.import.generatesMetadataFilesAtProjectRoot=false -DDetectVMInstallationsJob.disabled=true -Dfile.encoding=utf8 -XX:+UseParallelGC -XX:GCTimeRatio=4 -XX:AdaptiveSizePolicyWeight=90 -Dsun.zip.disableMemoryMapping=true -Xmx1G -Xms100m -Xlog:disable -javaagent:c:\\Users\\Public\\wpilib\\2024\\vscode\\data\\extensions\\redhat.java-1.33.0-win32-x64\\lombok\\lombok-1.18.33.jar -jar c:\\Users\\Public\\wpilib\\2024\\vscode\\data\\extensions\\redhat.java-1.33.0-win32-x64\\server\\plugins\\org.eclipse.equinox.launcher_1.6.900.v20240613-2009.jar -configuration c:\\Users\\Public\\wpilib\\2024\\vscode\\data\\user-data\\User\\globalStorage\\redhat.java\\1.33.0\\config_ss_win -data c:\\Users\\Public\\wpilib\\2024\\vscode\\data\\user-data\\User\\workspaceStorage\\7e3b38292c18d351fd0ed951ec504edb\\redhat.java\\ss_ws',
  level: 'info',
  timestamp: '2024-08-09 22:46:41.579'
}
{
  message: 'Starting Java server with: C:\\Users\\Public\\wpilib\\2024\\jdk\\bin\\java --add-modules=ALL-SYSTEM --add-opens java.base/java.util=ALL-UNNAMED --add-opens java.base/java.lang=ALL-UNNAMED --add-opens java.base/sun.nio.fs=ALL-UNNAMED -Declipse.application=org.eclipse.jdt.ls.core.id1 -Dosgi.bundles.defaultStartLevel=4 -Declipse.product=org.eclipse.jdt.ls.core.product -Djava.import.generatesMetadataFilesAtProjectRoot=false -DDetectVMInstallationsJob.disabled=true -Dfile.encoding=utf8 -XX:+UseParallelGC -XX:GCTimeRatio=4 -XX:AdaptiveSizePolicyWeight=90 -Dsun.zip.disableMemoryMapping=true -Xmx1G -Xms100m -Xlog:disable -javaagent:c:\\Users\\Public\\wpilib\\2024\\vscode\\data\\extensions\\redhat.java-1.33.0-win32-x64\\lombok\\lombok-1.18.33.jar -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath=c:\\Users\\Public\\wpilib\\2024\\vscode\\data\\user-data\\User\\workspaceStorage\\7e3b38292c18d351fd0ed951ec504edb\\redhat.java -Daether.dependencyCollector.impl=bf -jar c:\\Users\\Public\\wpilib\\2024\\vscode\\data\\extensions\\redhat.java-1.33.0-win32-x64\\server\\plugins\\org.eclipse.equinox.launcher_1.6.900.v20240613-2009.jar -configuration c:\\Users\\Public\\wpilib\\2024\\vscode\\data\\user-data\\User\\globalStorage\\redhat.java\\1.33.0\\config_win -data c:\\Users\\Public\\wpilib\\2024\\vscode\\data\\user-data\\User\\workspaceStorage\\7e3b38292c18d351fd0ed951ec504edb\\redhat.java\\jdt_ws',
  level: 'info',
  timestamp: '2024-08-09 22:46:42.830'
}
{
  message: 'WARNING: Using incubator modules: jdk.incubator.foreign, jdk.incubator.vector\r\n',
  level: 'info',
  timestamp: '2024-08-09 22:46:43.229'
}
{
  message: 'Aug 09, 2024 10:46:47 PM org.apache.aries.spifly.BaseActivator log\r\n' +
    'INFO: Registered provider ch.qos.logback.classic.servlet.LogbackServletContainerInitializer of service jakarta.servlet.ServletContainerInitializer in bundle ch.qos.logback.classic\r\n',
  level: 'info',
  timestamp: '2024-08-09 22:46:47.543'
}
{
  message: 'Aug 09, 2024 10:46:47 PM org.apache.aries.spifly.BaseActivator log\r\n' +
    'INFO: Registered provider ch.qos.logback.classic.spi.LogbackServiceProvider of service org.slf4j.spi.SLF4JServiceProvider in bundle ch.qos.logback.classic\r\n',
  level: 'info',
  timestamp: '2024-08-09 22:46:47.545'
}
```
Here's the java extension log file.

sciencewhiz, 2024-08-10:
That doesn't show anything wierd.

Can you try setting `"java.configuration.detectJdksAtStart": false,` in `C:\Users\Public\wpilib\2024\vscode\data\user-data\User\settings.json`

rnadu, 2024-08-13:
Tried it and it didn't change the results or error message.

sciencewhiz, 2024-10-28:
Try adding this to `C:\Users\Public\wpilib\2024\vscode\data\user-data\User\settings.json`

```
  "java.configuration.runtimes": [
    {
      "name": "JavaSE-17",
      "path": "c:\\Users\\Public\\wpilib\\2024\\jdk"
    },
  ]
```

closed 2024-10-31 by PeterJohnson