wpilibsuite/allwpilib #6845: 2026: deprecate robotInit

spacey-sooty, 2024-07-17:


brettle, 2024-07-18:
Will `simulationInit()` also be deprecated?

calcmogul, 2024-07-18:
It can be replicated with an `isSimulation()` if statement in the constructor, but `simulationInit()` doesn't feel as redundant as `robotInit()`.

brettle, 2024-07-18:
Since you can't override constructors in anonymous subclasses, loss of `simulationInit()` would preclude the following code:

```java
var testRobot1 = new MyTimedRobot() {
  public void simulationInit() {
    // test-specific robot initialization code
  }
};
assertRobotWorks(testRobot1);
```

which is convenient for testing. To workaround the loss of `simulationInit()` in this context one would need to create named subclasses or override `simulationPeriodic()` to contain some code that disables itself after the first call. Not a huge deal but something to consider.

Starlight220, 2024-07-18:
You can't override constructors, but you can make an initialization block which fulfills basically the same purpose

brettle, 2024-07-18:
Fair enough, but an initialization block still isn't quite the same as `simulationInit()`. `simulationInit()` is called by `TimedRobot.startCompetition()` (which in my example would be called during `assertRobotWorks()`), so it is not run at the same time as a constructor or initialization block.

calcmogul, 2024-07-19:
simulationInit() being called in startCompetition() instead of the robot constructor doesn't actually matter. robotInit() is called right after the robot constructor, and simulationInit() is called right after robotInit(). See https://github.com/wpilibsuite/allwpilib/pull/6623#issuecomment-2121081077 for more.

brettle, 2024-07-22:
`startCompetition()` (and thus `robotInit()/simulationInit()`) is called right after the robot constructor *if* the robot is run via `startRobot()`, but in my example above `assertRobotWorks()` doesn't call `startRobot()`. It calls `startCompetition()` with the robot that was constructed earlier. The `simulationInit()` hook makes it easier to write tests where the simulation initialization needs to change in a way which depends on something that happens between when the robot is constructed and when `startCompetition()` is called (e.g. something in `assertRobotWorks()` above). That said, if both `robotInit()` and `simulationInit()` are removed, one could achieve the same effect by overriding `startCompetition()` to run the code that would have been in them and then call `super.startCompetition()`. So I no longer see a downside to removing `simulationInit()` when `robotInit()` is removed.

calcmogul, 2024-07-22:
An initializer block as suggested in https://github.com/wpilibsuite/allwpilib/issues/6845#issuecomment-2237273666 should meet that need. From https://docs.oracle.com/javase/tutorial/java/javaOO/initial.html:

> Initializer blocks for instance variables look just like static initializer blocks, but without the static keyword:
>
> {
>     // whatever code is needed for initialization goes here
> }
>
> The Java compiler copies initializer blocks into every constructor. Therefore, this approach can be used to share a block of code between multiple constructors.

brettle, 2024-07-22:
> Just make a new object for every test with a different initialization block.

While I agree that would often be sufficient, my point was just that an initialization block is not _technically_ equivalent to `simulationInit()` because other code can be run between when the initialization block is run and when `simulationInit()` is run. The actual equivalent to `simulationInit()` (once `robotInit()` and `simulationInit()` have been removed) would be code inserted at the beginning of an overriden `startCompetition()`. Since that equivalent would exist, I'm fine with the removal of `robotInit()` and `simulationInit()`).

Just for clarity, as an example of where an initialization block would not be equivalent to `simulationInit()` and switching to using one might not be preferred, consider:

```java

MyTimedRobot testRobot;

boolean testRequiresSpecialSim;

@BeforeEach
void setup() {
  testRobot = new MyTimedRobot() {
    public void simulationInit() {
      if (testRequiresSpecialSim) {
        initSpecialSim();
      }
    }
  };
}

@Test 
void normalTest() {
  assertRobotWorks();
}

@Test 
void specialTest1() {
  testRequiresSpecialSim = true;
  // ... other test-specific test setup
  assertRobotWorks();
}

@Test 
void specialTest2() {
  testRequiresSpecialSim = true;
  // ... other test-specific test setup
  assertRobotWorks();
}

void assertRobotWorks() {
  testRobot.startCompetition();
  assertGotExpectedResult1();
  // ...
  assertGotExpectedResultN();
}

```

But again, I'm ok with the removal of `robotInit()` and `simulationInit()` because in that case I could get the same result by just changing the `setup()` method to be:

```java
void setup() {
  testRobot = new MyTimedRobot() {
    public void startCompetition() {
      if (testRequiresSpecialSim) {
        initSpecialSim();
      }
      super.startCompetition();
    }
  };
}

```

calcmogul, 2024-07-22:
I'm actually not sure what test-specific stuff you'd even put in a test-only anonymous class. The simulation init stuff _should_ already be in the robot class or subsystems.

brettle, 2024-07-22:
As a specific example, perhaps for the "special" tests we want to use an external virtual world simulator instead of the sims created in the robot class. In that case, `initSpecialSim()` might be responsible for setting that up, including connecting to the external simulator. But the external simulator might not be started until just before calling `startCompetition()` in  `assertRobotWorks()`.

To be clear, I agree that there are other ways to achieve this. I'm just providing an example of a way that someone might want to organize their code when using `simulationInit()` that is not equivalent to having that code in an initialization block. Hope that helps.

calcmogul, 2024-07-28:
Duplicate of #6622.

closed 2024-07-28 by calcmogul