wpilibsuite/allwpilib #6860: [commands, documentation] Clear out Commands examples.

spacey-sooty, 2024-07-20:
The current commands examples are inconsistent and mostly do not align with best practice. I believe this is due to a lack of support for all of the different examples as there are so many. The examples should be cut down and narrowed down to one or two which are kept up to date with current best practice command based reccomendations. A good example for this would be something like gears bot as it covers a lot of features.

Starlight220, 2024-07-20:
RapidReactCommandBot is often referred to as an up-to-date example.

spacey-sooty, 2024-09-08:
https://github.com/wpilibsuite/allwpilib/pull/7054 and https://github.com/wpilibsuite/allwpilib/pull/7053 are good steps in the right direction along with https://github.com/wpilibsuite/frc-docs/pull/2684