wpilibsuite/allwpilib #6896: [Epilogue] Add support for persistant variables

scarmain, 2024-07-29:
**Is your feature request related to a problem? Please describe.**
Currently, we use NT's persistent feature for calibration values like swerve offsets, max speeds, pid constants, etc.  It would be nice if we can use Epilogue with an annotation to mark what variables should be saved in the NT persistent table.

**Describe the solution you'd like**
A annotation like @Persistent to mark that a field is persistent.  Maybe it needs a default parameter too, so like @Persistent(default=42)

**Describe alternatives you've considered**
I can go back to the old preferences API, but would prefer not to.  (https://docs.wpilib.org/en/stable/docs/software/basic-programming/robot-preferences.html#setting-robot-preferences)

**Additional context**
I think Epilogue is super awesome and I'm super excited to try it in the robot code next season!!!

rzblue, 2024-08-01:
Could you elaborate on what how this would behave and your use cases? Since Epilogue is a logging library, values cannot be read from it, only logged with it. Persistent NT values don't really make sense here.

scarmain, 2024-08-04:
I agree with your observation right now.  I didn't look close enough to the logging method, and realize it's the straight USB file based, not NetworkTables to logging.  There would be no point as is to add persistence currently.

I guess, seeing this now, is there a reason why none of these values are not on NetworkTables too?  If I'm prototyping/testing something, it makes a lot more sense to have the data on NT instead; that way you can see it live with the robot's actions.  This way of logging will force you to do a run, pull the flash drive, make a change, replace flash drive, and repeat.  Maybe we could meet in the middle and you can annotate fields you do want on NT too.  (then adding persistence is just .setPersistent(true) )

Gold856, 2024-08-04:
Epilogue does output to NetworkTables as well as DataLog (the outputs are configurable). However, this isn't really what Epilogue was made to do. Epilogue won't read the value into the field for you, it only logs, so changes you make on NT won't reflect back to the variable.

At this point, why not use a Constants class? It's slightly slower than using NT persistence since you have to deploy to make a change, but the advantage is that the constants stick to the code, which means it gets tracked in version control and is harder to lose.

scarmain, 2024-08-04:
1) When you have multiple robots, each robot can remember it's constants, instead of having to make multiple constant files.
2) You can see calibration constants change live, so you can be doing work without needing a programmer there too.
3) Some constants change with drivers or environment, like maximum speed for safe operation in an compact area like the practice fields or demos.

We have our reasons for using them, and I'd prefer not to debate them here.  Live tuning of constants is a great feature, and I wish that WpiLib would continue to support it.

PS: I do admit there are issues with persistent tables too, if you do a Rio swap, the data from the networktables.json doesn't come with it, and there is no easy way to move it (scp it from one Rio to another).  We have also had the file dropped in the past (once we noticed on a competition field before a match, once in a practice in January), so we do back up this file on occasion.  I've thought about making an issue to improve this (since there is a web handler to download the config (http://10.28.32.2:5810/nt/persistent.json), why can't we make a simple HTML page to upload it also).

ThadHouse, 2024-08-04:
"Epilogue won't read the value into the field for you, it only logs, so changes you make on NT won't reflect back to the variable." This is still a key issue with doing this. Epilogue is purely for logging, and is one way. It never writes fields. Its actually a fairly major architectural change to have it write fields in addition to read.

Now a persistent annotation API might be a nice thing, but it really wouldn't be epilogue. It'd be its own thing, because it'd be doing something completely different, with basically no relation to logging itself.

KangarooKoala, 2024-08-05:
> **Describe alternatives you've considered**
> I can go back to the old preferences API, but would prefer not to.

Could you elaborate more on why you would prefer not to use the old preferences API? Maybe it'll be better to improve that rather than trying to changing Epilogue's purpose.

scarmain, 2024-08-06:
As I said above, I thought Epilogue would convert our member variables to NT publishers, so adding persistence would have been easy.  As this discussion as brought up, Epilogue instead uses DataLog objects to store all the data, so this persistence request will not work.  As I said above, it would have been nice if this data was instead on NT and logged with NT logging, as we could debug issues live, instead of using a flash drive.  I should probably close this issue as a bad idea (I agree it is knowing how Epilogue works more), but there is still a persistence conversation going on too.

-------

@KangarooKoala Ironically, we don't use the Preferences API.  I find it clunky the way it is setup.  You shouldn't have to Init the variable, and as part of init, you should pass in the default as part of initialization.  (I also don't know if you can subkey with that strategy, like pass in "Shooter/kP" to organize calibrations)  

We actually created our own version of Preferences that is more NT4 styled.  Like this [one](https://github.com/FRC2832/Robot2832-2024/blob/a62768eecd1c5c09e2e80156c8081e2b5a64f3c8/src/main/java/org/livoniawarriors/UtilFunctions.java#L106), you just call our UtilFunctions.getSettingSub(key, default), then it returns a DoubleSubscriber to .get() the value whenever you want.  

We also have helper functions in there for simple Publishers to NT4, as I find implementing the NT4 API cumbersome instead of SmartDashboard.putNumber() that WpiLib is trying to discourage using (both old NT3 style of connection and the SmartDashboard prefix attached to the key).  For the NT4 interfaces, you have to find a NT table instance, create a topic, then publish it, set a default, then use it normally.  Our [helper function](https://github.com/FRC2832/Robot2832-2024/blob/a62768eecd1c5c09e2e80156c8081e2b5a64f3c8/src/main/java/org/livoniawarriors/UtilFunctions.java#L234) does all that so we don't have to constantly repeat all those steps.

rzblue, 2024-08-08:
> As I said above, I thought Epilogue would convert our member variables to NT publishers, so adding persistence would have been easy. As this discussion has brought up, Epilogue instead uses DataLog objects to store all the data, so this persistence request will not work. As I said above, it would have been nice if this data was instead on NT and logged with NT logging, as we could debug issues live, instead of using a flash drive. I should probably close this issue as a bad idea (I agree it is a bad idea now knowing how Epilogue works more), but there is still a persistence conversation going on too.

Epilogue *does* log to NetworkTables, if the user enables it. But it's *only* logging (publishing) to NetworkTables- it does not bind variables to NetworkTables topics bidirectionally (ie, if another client publishes data to a topic being logged to by Epilogue, that data is not reflected in the variable. That's why persistence doesn't make sense here.

scarmain, 2024-08-13:
Closing due to not implementable

closed 2024-08-13 by scarmain