wpilibsuite/allwpilib #6919: Add usage reporting ID for the LEDPattern API

SamCarlberg, 2024-08-02:


sciencewhiz, 2024-12-02:
How do you see the usage being reported? Since multiple patterns can be used for different parts of the strip, it doesn't make sense to do instances for the different patterns since only the last would be reported. There doesn't look to be a singular place to report a LEDPattern Resource, and I don't think it makes sense to make a bunch of resources for each pattern. Count of the number of times any LEDPattern is created (which would have to be done throughout the class)?

SamCarlberg, 2024-12-02:
The factory and modifier methods could each report a resource, or just report a general API usage when the LEDPattern class is loaded in a static initializer. The former is more granular, but means maintaining a largish list of resource IDs and can't detect custom user-defined patterns.

auscompgeek, 2024-12-02:
> report a general API usage when the LEDPattern class is loaded in a static initializer

This approach won't work for C++ and hence Python, so I think we'll have to settle for the former at least for C++.

closed 2025-12-05 by PeterJohnson