wpilibsuite/allwpilib #6921: Remove empty defaults from switch statements and expressions

spacey-sooty, 2024-08-03:


rzblue, 2024-08-04:
In java switch statements on enums are not checked for exhaustiveness, and non-enum values can't be checked for exhaustiveness. It's generally considered good practice to have a default to make it clear what happens with all other values. Switch statements obviously must be exhaustive, but for non-enum values you need a default case.

Is this for c++ specifically?

spacey-sooty, 2024-08-05:
No this is also for Java, by adding an empty default to a switch expression you add absolutely nothing to the expression. If you were to instead add a default which does something (eg logs an invalid state) this would make sense but this is not what WPILib does right now.

rzblue, 2024-08-08:
Perhaps instead of removing all default cases we should throw exception or log a warning where appropriate, or add a comment indicating why it's ok that the default was reached. IMO not having a default is only fine where it's obvious why unhandled cases can be safely ignored.

>  by adding an empty default to a switch expression you add absolutely nothing to the expression.

Are you referring to a statement here instead? Expressions can't have empty cases