wpilibsuite/allwpilib #6939: Memory Leek When Robot Container Object is Deconstructed

Jetvac2, 2024-08-08:
**Describe the bug**
When the when Robot Container object is deconstructed and reconstructed multiple times in the same robot boot cycle/code deploy eventually(after 4-5 times) the following error is outputted and the code crashes. The memory leak also appears to happen during simulation as the simulation speed slows down the more time the robot container is  deconstructed and reconstructed but I have never bean able to make my PC crash do to an out of memory error. 
"[519.055s][warning][os,thread]Failed to start thread "Unknown thread" -pthread_create failed(EAGAIN) for attributes: stacksize 320k, guardsize: Ok, detached.
"[519.055s][warning][os,thread] Failed to start native thread for java.lang.Thread "Thread-6"
Unhandled exception: java.lang.OutOfMemoryError: unable to create native thread: possibly out of memory or process/resource limits reached. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new RobotProject
2. In the teleopInit method add "this.robotContainer = new RobotContainer();"
3. Instead of 2 add "this.robotContainer = new RobotContainer();" to the disableInit method
4. Enable and disable the robot until it crashes

 - Link to code:
https://github.com/ZetaBotics2024/Team7826Lib/blob/main/src/main/java/frc/robot/Robot.java

**Expected behavior**
The RoboRio should not run out of memory. 

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Project Information: WPILib Information:
Project Version: 2024.3.2
VS Code Version: 1.85.1
WPILib Extension Version: 2024.3.2
C++ Extension Version: 1.19.1
Java Extension Version: 1.26.2023121408
Java Debug Extension Version: 0.55.2023121302
Java Dependencies Extension Version 0.23.2023120100
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    AdvantageKit (3.2.1)
    ChoreoLib (2024.2.3)
    PathplannerLib (2024.2.8)
    CTRE-Phoenix (v5) (5.33.1)
    CTRE-Phoenix (v6) (24.3.0)
    photonlib (v2024.3.1)
    REVLib (2024.2.4)
    WPILib-New-Commands (1.0.0)

**Additional context**
This has only bean tested on the RoboRio1

PeterJohnson, 2024-08-08:
This is not surprising, as RobotContainer looks like it allocates a lot of things and doesn’t have a close() or similar to ensure everything is released.  Also, Java lazily recycles no longer referenced memory, so it may take a while to release allocated objects.

Why are you creating RobotContainer multiple times?

Jetvac2, 2024-08-08:
For some reason I can not deduce when I run an auto multiple times on the same robot code cycle it produces increasing erratic behavior for every run after the first one. So as a quick cover up solution to allow for easier auto tuning we simply reconstruct the robot container every time the robot disables after having run an auto. This is obviously a bad solution to the issue but it has suited or needs so far. I just noticed that this would make the robot crash if this was done enough times and figured I would report the issue incase it was a symptom of a larger issue.

mcm001, 2024-08-08:
If you properly close all resources allocated by your old RobotContainer, does the memory usage still climb?

Jetvac2, 2024-08-08:
I don't know. I am using java so I would imagine that all of the object created within the robot container should be automatically freed when the container is deconstructed. I am mistaken?

ThadHouse, 2024-08-08:
No, Java does not work like that. When no references are left to the object container, it might get cleaned up sometime in the future by the garbage collector. But there is no deterministic cleanup, unless you explicitly close all resources used by RobotContainer and everything contained inside of RobotContainer.

Jetvac2, 2024-08-09:
Ok, I did not know that. Thanks for letting me know.

mcm001, 2024-08-09:
> No, Java does not work like that. When no references are left to the object container, it might get cleaned up sometime in the future by the garbage collector. But there is no deterministic cleanup, [...]

Bonus points for the GC not knowing how large native objects owned by Java classes (like image mats) are, so it doesn't realize all these tiny Java classes are actually causing tons of memory pressure

closed 2024-09-13 by calcmogul