wpilibsuite/allwpilib #6991: [wpilibc] Check for invalid handles in destructors

PeterJohnson, 2024-08-22:
Moved objects invalidate the moved-from hal::Handle.  We shouldn't call the HAL free handle function in this case, and certainly shouldn't print an error.

Potentially the hal::Handle class could be improved to add a template parameter for the HAL free handle function so this could be refactored?  This might even enable destructors to be default'ed in a number of cases.

rzblue, 2024-08-23:
Adding a template parameter to the wrapper was easy enough, but you can't instantiate the template without access to the free function, and we don't want to include the full HAL header for each type. Any ideas?

PeterJohnson, 2024-08-24:
Yeah, that makes sense.  Is there a major downside to including the HAL header though?  Since it’s just C function declarations those headers are quite lightweight.

rzblue, 2024-08-24:
It's not heavy, it just adds more HAL symbols to the user's global namespace. We could wrap the include in a `detail` namespace - the functions are extern C so it would work.

PeterJohnson, 2024-08-24:
They're all prefixed with HAL_, so it shouldn't be an issue to do that?

rzblue, 2024-08-24:
It feels a little icky to me, but there's nothing inherently wrong with it. If you're ok with it then let's just include it.

PeterJohnson, 2024-08-26:
Yes, I'm fine with it.  We do the same thing many other places (e.g. the NT C++ header includes the NT C header).

PR #7016 (2024-09-07): [hal] Check for invalid handle before closing handles
Closes #6991 

Adds a close function pointer template parameter to `hal::Handle`- this allowed default destructors in many places. 
The status parameter has been removed from close functions- in most places it was not used. Where it was, an error is printed in stead.
JNI functions also check for invalid handle- though now that i'm thining about it, I'm not sure this is really necessary. It's easy enough to remove.
closed 2024-09-07 by PeterJohnson