wpilibsuite/allwpilib #7036: Make wpi::MemoryBuffer::GetFile() use wpi::expected

calcmogul, 2024-09-05:
The current MemoryBuffer API has a verbose side channel for error handling, and it's unclear from the API itself what errors need to be handled, and in what way:
```cpp
std::error_code ec;
if (auto file = wpi::MemoryBuffer::GetFile("file.txt", ec);
  file == nullptr || ec) {
  // Do something with ec
  return false;
} else {
  file->GetCharBuffer();
}
```
I'd rather have an API like `wpi::expected<wpi::MemoryBuffer, std::error_code> GetFile(std::string_view)`:
```cpp
if (auto file = wpi::MemoryBuffer::GetFile("file.txt"); file) {
  file.value()->GetCharBuffer();
} else {
  // Do something with std::error_code in file.error()
  return false;
}
```

PeterJohnson, 2024-09-05:
Other file APIs would also benefit, although a lot of them use constructors rather than factory functions, so that would be a more major change.

KangarooKoala, 2024-09-07:
Would we keep the existing overload, deprecate it, or remove it?

calcmogul, 2024-09-07:
We can just remove it since it's basically an internal API for us.

PR #7069 (2024-09-13): [wpiutil] Use wpi::expected in MemoryBuffer::GetFile
Closes #7036.

I'm not completely happy with the idiom I went with, but I think it works relatively well. If anyone has suggestions for a cleaner one, I'd be happy to hear it!

Also, I'm a little worried about how long the `GetFile()` declaration line is in the header file, but it's excluded from format so I'm not sure what to do.
closed 2024-09-13 by PeterJohnson