wpilibsuite/allwpilib #7094: Document what the rumble types mean

rzblue, 2024-09-19:
On Xbox controllers, left is the low frequency motor and right is the high frequency motor. Not sure if PS4 is the same.

https://learn.microsoft.com/en-us/gaming/gdk/_content/gc/reference/input/xinputongameinput/structs/xinput_vibration

TheTripleV, 2024-09-19:
Well its weird. On some controllers, there's actually 2 motors, distinctly left and right with no frequency difference

ThadHouse, 2024-09-19:
PS4 doesn't work anyway. setRumble only works for XInput controllers. As for left and right, the official are high and low frequency. But 3rd party ones might not follow that.

oh-yes-0-fps, 2024-09-19:
Using ds4windows allows rumble to work on ps4/ps5 controllers and it behaves the same as xbox