wpilibsuite/allwpilib #7108: [cscore] Replace OpenCV with SIMD library

PeterJohnson, 2024-09-20:
OpenCV is a large dependency for cscore, but outside of interfaces is only used for image storage, resize, color conversion, and JPEG compression/decompression.  The SIMD library (https://github.com/ermig1979/Simd) has all of these capabilities with accelerated support for NEON (ARM) and desktop platforms.  Replacing OpenCV with SIMD for internals would enable cscore to be built without OpenCV, enabling easier platform builds and language use (e.g. Python).

gerth2, 2024-10-10:
[image]

gerth2, 2024-10-10:
But in all seriousness. The size/complexity of open CV was a big contributor to the difficulties photonvision had in getting the robotpy wrappering mechanism to work out of the box on the C++ vendordep.

The ability to overlay shapes or edit individual pixels in an image (IE, get camera image, annotate it, then stream it) is still useful. But... I"m assuming SIMD has some concept of "set pixel x/y to (value)"? Even if text/shape drawing had to be reimplemented on top of that, it still seems like a viable tradeoff.

PeterJohnson, 2024-10-10:
The goal here is simply to remove the dependency for cscore, not replace opencv functionality completely.  We could still have opencv available at a higher level of integration.