wpilibsuite/allwpilib #7123: Implement Tsitouras 5th order numerical integrator

calcmogul, 2024-09-23:
This issue is a spin-off of https://github.com/wpilibsuite/allwpilib/issues/4373.

The Tsitouras 5th order integration method is 3x more accurate than Dormand-Prince (RKDP). It uses the exact same algorithm with different coefficients.

Section 7.9 of [Controls Engineering in FRC](https://file.tavsys.net/control/controls-engineering-in-frc.pdf) has an introduction to the notation of Runge-Kutta methods.

Here's the paper to implement: http://users.uoa.gr/~tsitourasc/RK54_new_v2.pdf

Here's the implementation steps:

1. Copy-paste the RKDP function(s) to make a new function Tsit5
2. Update the coefficients of A and b to reflect those in section 3, table 1 of the paper above
3. Update the source location comment in the function
4. Copy-paste and modify the tests
5. Replace usages of RKDP with Tsit5

Advay17, 2024-11-12:
If RKDP usages are replaced, should it be deprecated, or are there any other use cases for it?

calcmogul, 2024-11-12:
RKDP should be deprecated since Tsit5 is superior in every way.