wpilibsuite/allwpilib #7124: Privacy Policy

EndlessEevee, 2024-09-23:
My school requires a privacy policy to use wplib. Do you have one?

calcmogul, 2024-09-23:
WPILib itself doesn't collect any user data for a privacy policy to apply to. FIRST's Field Management System collects usage reporting data at competition, which would probably be governed by https://www.firstinspires.org/about/privacy-policy.

sciencewhiz, 2024-09-24:
Can you share more details about the schools requirements? Does the mere existence of a privacy policy suffice, or are there certain things the school is looking for? Does this cover software installed on school owned computers, or software used in school sanctioned activities, or what?

EndlessEevee, 2024-09-24:
The school requires a policy that states everything the app or developers may collect. This is required for all apps to be installed in a school device. Sadly that is all I know

EndlessEevee, 2024-09-24:
Something like this would suffice. https://github.com/Gold872/elastic-dashboard/blob/main/PRIVACY.md

sciencewhiz, 2024-09-24:
> The school requires a policy that states everything the app or developers may collect.

They don't care what data is collected, just that it's disclosed?

sciencewhiz, 2024-09-24:
> The school requires a policy that states everything the app or developers may collect.

So it doesn't matter what data is collected, as long as it is disclosed?

EndlessEevee, 2024-09-24:
While they do care about what data is collected, they just need to have something to go off of.
Mason

closed 2024-09-28 by jasondaming