wpilibsuite/allwpilib #7130: Add physics simulation classes with torque input instead of voltage input

calcmogul, 2024-09-25:
**Is your feature request related to a problem? Please describe.**
Current and torque control are becoming popular in CTRE's ecosystem, and our physics sim classes that use a voltage input aren't compatible with it.

**Describe the solution you'd like**
The current class is hardcoded to use a voltage input, so it's cleanest to have separate classes for each model type. Of the two choices, current or torque, the latter is more generalizable to all kinds of mechanisms.

The torque input should be the direct actuator output, not what the mechanism experiences directly. Thus, the model should include a gear ratio to convert between the two.

**Describe alternatives you've considered**
We could make all physics sim classes take a torque input, with add-ons for the motor dynamics to adapt voltage inputs or current inputs to torque inputs.

narmstro2020, 2024-10-03:
I'd like to take a stab at this.

narmstro2020, 2024-12-18:
I opened a PR on this awhile back, but I'm sure things have gotten busy and I'm confident this is a low priority feature.  

Some thoughts I'd like to pick your brains on.  

CTRE allows for switching controls modes during operation.  

Should the simulation classes take that into account?  

Also it doesn't really seem doable to allow for this with one DCMotor input object.  The two (soon 3) CTRE motors have different DCMotor objects.  It feels like each sim class would have to be based off of 2 LinearSystemSim that keep their states in sync.  

Or create a completely new base class for this sim type.

calcmogul, 2024-12-18:
We'll have to think about physics sim more holistically instead of bolting on some torque classes. That makes it more of a 2027 task.