wpilibsuite/allwpilib #7135: Console output is garbled in datalogs

rzblue, 2024-09-28:
The console output added in #6977 is garbled (seems to be missing lots of characters):

```
1.528  | Wbi 2 on port 0 not available, check if controller is plugged in
11.914 | **** Rbo rga tru omlt ******Wrig t d.p.istwiDrvrtto.rprJysicUnlgeWarig(rverSaio.aa146:Jysik uton  npr 0ntaalbe hekifcnrlle spugdinWn tpilb.IaieooBs.tLoOeruMsag(Ieaieotaejaa:3):Lo if .Wrigedfrt.rae.aEs$0a6)ahords(:046s	dsaIit):0022s	otProdi(:0099	LvWino.pdatVle(: .0s	hffeor.pdae): .003
11.915 | diabedeidc) .042
12.607 | Wariga duwifrtwiij.rvertto.eotosiknlggedann(rveSaio.jav:14 osikBton 2 npr  o viale, hc fcnrleri luggi
15.650 | Wariga duwifrtwbjDitnotosiknleWrigDivrttonjv:34) osikBto  o o ntalal,chc f otori lge nWring teuwifrt.piij.rvrtt.rprJytcknlgedanDieSain.aa14JtBtn2 npr  ot via cek ifcnrleri lge nWanneps.plbrieStation.reportJoystickUnpludWarning(DriverStation.java:1346): Joystick Button 2 on prt 0 notavailable, heck if controller is plugged i
16.650 | Warning at edu.wpi.first.wpilibj.DriverStation.reportJoyckUnpluggedWarning(DriverStion.java:1346): Joystick Button 2 on port 0 not avaiable, heck if controler is pluged n
17.671 | Wrning at edpi.first.wpilibj.DriverStation.repJoystickUnpluggedWaring(DriverStation.java:1346): Joystick Button 2 on port 0 nt avalable, check if controler is pluged i
18.690 | Warning ..frst.wpilibj.DriverStation.reportJotickUnpldWarning(DriverStation.java:1346): Joystick Button 2 o port0 not available, check i controler is plugged in
19.691 | Warning at edu.first.wpilibj.DriverStation.rrtJoystickUnpluggedWarning(DriverStation.jav1346): Joystick Button 2on port 0 not available, check if ontroller i plugged in
20.710 | Warning at edu.wpi.firspilibj.DriverStation.reportJipledWarning(DriverStation.ava:1346): Joystick Button 2 on port 0 not availabl, check if controlle is plugged n
21.730 | Wrning at edu.wpi.fiwiDrirSation.reportJoystickUnpluggedWarni(DriverStation.java:134): Joystick Button 2 on port 0 not available, check if controller is plugged in
22.748 | Wann teuwifrtwiijDieSain.reportJoystickUnpluggedWarning(DriverStation.java:1346):oystick Button 2 on port 0 notavailable, ceck if controlle is pluged in
```

Should be:
```
********** Robot program starting **********
NT: Listening on NT3 port 1735, NT4 port 5810
DataLog: Logging to '/u/logs/FRC_TBD_2624151fddf7ef7c.wpilog' (28.8 GiB free space)
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
********** Robot program startup complete **********
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
Warning at edu.wpi.first.wpilibj.IterativeRobotBase.printLoopOverrunMessage(IterativeRobotBase.java:436): Loop time of 0.02s overrun

Warning at edu.wpi.first.wpilibj.Tracer.lambda$printEpochs$0(Tracer.java:62):   SmartDashboard.updateValues(): 0.003469s
        disabledInit(): 0.002727s
        robotPeriodic(): 0.029291s
        LiveWindow.updateValues(): 0.000111s
        Shuffleboard.update(): 0.000093s
        disabledPeriodic(): 0.000482s

Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
Warning at edu.wpi.first.wpilibj.DriverStation.reportJoystickUnpluggedWarning(DriverStation.java:1346): Joystick Button 2 on port 0 not available, check if controller is plugged in
```

(the first 3 lines occur before datalog starts, so those are expected to be missing from the output)

spacey-sooty, 2024-09-28:
It appears whitespace characters like \n are being excluded as \ and n.

rzblue, 2024-10-02:
Newlines should be getting stripped from the output, it should be outputting one log entry per line. What do you mean by "getting excluded as \ and n"?

The missing characters aren't newlines, they're just characters in the middle of a line. I have a few ideas what it could be, just need to debug it on a rio.

PR #7150 (2024-10-08): [wpiutil] Refactor FileLogger
FileLogger now logs entire flushed blocks of data, instead of breaking it up into lines.

Fixes #7135.
closed 2024-10-08 by PeterJohnson