wpilibsuite/allwpilib #7146: Networktables server keeps running after closing sim

DanPeled, 2024-10-02:
**Describe the bug**
Ive had a couple of times (more than a handful), the networktables to keep running after i closed the simgui completley and shut down vscode, not sure what may cause that or if i had any special stuff going on that may cause that, this issue has happend throughout a few wpilib projects ive worked on

**Desktop (please complete the following information):**
   - OS: Ubuntu 22.10
   - Project Information: WPILib Information:
Project Version: 2024.3.2
VS Code Version: 1.85.1
WPILib Extension Version: 2024.3.2
C++ Extension Version: 1.19.1
Java Extension Version: 1.36.2024092808
Java Debug Extension Version: 0.55.2023121302
Java Dependencies Extension Version 0.23.2023120100
Java Version: 17
Java Location: /home/dan/wpilib/2024/jdk
Vendor Libraries:
    AdvantageKit (3.2.1)
    NavX (2024.1.0)
    PathplannerLib (2024.2.4)
    CTRE-Phoenix (v5) (5.33.1)
    CTRE-Phoenix (v6) (24.2.0)
    REVLib (2024.2.3)
    ReduxLib (2024.1.1)
    WPILib-New-Commands (1.0.0)
    YAGSL (2024.4.8.6)
(this is the project ive had this issue the most)

DanPeled, 2024-10-02:
Not running anything with threads

rzblue, 2024-10-02:
How do you know it's still running?

DanPeled, 2024-10-02:
When opening a dashboard (tried with both elastic & ascope), they can still connect and have data modified & read

PeterJohnson, 2024-10-02:
Unfortunately, there could be many causes across the many vendordeps that cause the program to hang instead of properly exiting.  If you're familiar with debugging with gdb, you can try to figure out where it's hanging yourself, or you can generate and post a coredump for us to look at, or post your project so we can try to reproduce locally.

To generate a core dump: `kill -11` of the Java process running the user program will generate a core dump (assuming resource limits allow it, e.g. `ulimit -c unlimited`).  Alternatively, you could run gdb, attach to the PID, and run generate-core-file.