wpilibsuite/allwpilib #7156: Java Unit types missing `mult(U)` functions

bhall-ctre, 2024-10-03:
It looks like all the Java unit types have `per(U)` functions, but many of them are missing generic `mult(U)` functions. For now, users can use `MultUnit.combine(a, b)`, but it would be nice if users could directly do something like `Volts.per(Rotation.mult(Second))` for all units.

SamCarlberg, 2024-10-04:
So, a caveat to this is type erasure (of course) with combinatory units and their subclasses. Something like this wouldn't be possible:

```java
public class Per<N extends Unit, D extends Unit> extends Unit {
  N mult(D denom) { ... }

  <Other extends Unit> MultUnit<? extends Per<N, D>, Other> mult(Other other) { ... }
}
```

Because `N` and `Other` both erase down to `Unit` and result in identical method signatures

But basic non-combinatory types like voltage and angle should be fine. My main concern would be lack of consistency, like having only some units support `mult` while others don't or only in a limited way.

bhall-ctre, 2024-10-04:
Given that `mult()` and `per()` are both used moreso to create new units rather than to perform dimensional analysis (which is more what I would expect from `times()` and `divide()`), I think that's a fair trade-off. `MultUnit.combine(a, b)` already doesn't do any dimensional analysis either.